mwittig/node-milight-promise #2: White bulb/stripe control

Oggymator, 2015-10-07:
Hi,
is it right, that I can control only RGBW bulbs/stripes?!
I see in the App "EVE" only the 4 zones of the RGBW controller.
Is it possible to Add also the 4 zones of the Single/Dual White controller?
thanks
Oggy

mwittig, 2015-10-07:
Hi Oggy,

as far as I know this should be possible as there are remote controls with 4-zones for white LEDs - see http://www.amazon.com/LEDENET%C2%AE-Control-Wireless-Dimmable-Controller/dp/B00I09J4P0
I am not sure what the "EVE" App is, but if you use the official control Apps (Milight (iOS) or Mi.Light (Android) you can select a control for the dual white 4-zone controller. See also http://www.milight.com/support#A2 and http://www.milight.com/support#A4
By using the different controls it should also be possible to set-up a mixed-mode configuration,e.g. RGBW on zone1 and DualWhite on zone4.  Hope this helps. Please keep me posted.

mwittig, 2015-11-01:
no further response

closed 2015-11-01 by mwittig

mwittig, 2016-01-02:
Release 0.0.4 - Added example for WW/CW bulbs/strips