mwittig/node-milight-promise #7: LimitlessLED RGBWW Saturation Control

mattjanssen, 2016-09-23:
Looking at the LimitlessLED strip controllers page, there's a new one (as of 2016) that says it'll run both the RGB chips _and_ the white chips at the same time, giving us saturation control.

Do you think the LimitlessLED WiFi Bridge v5.0 can handle RGB + white at the same time? And if so, would this library handle it? Otherwise we're waiting for the WiFi Bridge v6.0 I assume.

> 4 COLOR CHANNELS WRGB, 4 Colour RGBW LED Strip controller (with Saturation control – new FOR 2016!) – max.288W (72W per channel x4) (max. 6.0 Amps @ 12V per channel) this new WRGB controller is same as the 4 color controller above except this controller forces the remote to use its 4 zone buttons instead as 4 individual brightness controls to set the brightness of each white/red/green/blue independently, giving some awesome white saturation control. The way you link this new WRGB controller is using the master ON button on both the RGBW remote and the iOS App for the wifi bridge – the 4 zones on the remote control each of the colors and brightness individually.

Oggymator, 2016-09-23:
Have to wait for the new bridge, that will come next month.
The production of the "old" one ends this month.
So if anybody need it, buy quick.

The new bridge will have a complete new App.
So I don't know if this Library will work with it.

mwittig, 2016-09-23:
> Looking at the LimitlessLED strip controllers page, there's a new one (as of 2016) that says it'll run both the RGB chips and the white chips at the same time, giving us saturation control.

I think this will only be supported with the newest generation if RGBWW/CW bulbs. See
http://www.limitlessled.com/shop/rgbw-ww-cw-light-bulb/

> Do you think the LimitlessLED WiFi Bridge v5.0 can handle RGB + white at the same time? 

I think it works more-or-less the same way as with v4.0.  The major difference is that you have up to 1000 bulbs perzone with v5.0 in contrast to 100 per zone with v4.0. For sure, there are also other advances - let's see see :)

Page 6 of http://www.limitlessled.com/download/LimitlessLED_Wifi_Bridge_v4_Instructions_March2014.pdf
"Each Wifi Bridge can control up to 4 RGBW Groups of Bulbs, 4 Dual White Groups of Bulbs, and 1
Group of RGB Bulb."

http://www.limitlessled.com/shop/wifi-udp-receiver-bridge/
"Each WiFi Bridge does up to 9 zones
• 4 zones of up to 1000 bulbs per zone for White,
• 4 zones of up to 1000 bulbs per zone for RGBW Color,
• and 1 zone of up to 1000 per zone for RGB Color. "

> So I don't know if this Library will work with it.
I am not sure either, but I'll do my best to get hold of a v5.0 bridge as soon as possible and do the necesseary changes the Library. Most of this should work the same as for v4.0. The developers page is is already highlighting the "LimitlessLED v5.0 OpenSource API" for months now, but there isn't much information available, yet apart from a new version of the admin tool. I guess, there will be new control sequences the RGBWW/CW bulbs which give you saturation control. Let's see.
http://www.limitlessled.com/dev/

Oggymator, 2016-09-23:
The RGB+Dual White Buld(RGB+CCT) will not work with the Bridge V4.0 oder V5.0.
For this you need the complete redesigned Wifi Bridge that will come in the next month.

mwittig, 2016-09-23:
Thanks, for pointing this out! In the middle of writing I mixed-up the version numbers. I edited my previous post accordingly.

Oggymator, 2016-11-28:
The new Bridge is available, so satturation control by App is possible with the RGB+CCT controller or RGB+CCT lights.
Here the Link from Limitlessled with the API:
http://www.limitlessled.com/dev/

Maybe you can make a second plugin for the new bridge, I dont know what is changed to the old one.

Also you can buy it from me if you want(German reseller), but I will not make some promotion here. Only if mwittig will allow it.

mwittig, 2016-11-28:
Thanks for letting me know! Buying from you would also help to speed up things.  Can you please send me an email with details on pricing/shipping costs. Thanks!

See my coordinates at
https://www.npmjs.com/~mwittig

Oggymator, 2016-11-28:
send an email, maybe have to look in the spam folder ;)

rspaargaren, 2016-11-29:
Nice, just checking if  @mwittig was picking up the conversion to the v6 bridge. As I have this box in my house now and love to see the integration with Pimatic working.

rspaargaren, 2016-11-30:
on the miligth tweakers forum more info can be found on how to interpretate the limitlessled dev information.

mwittig, 2016-12-01:
@rspaargaren  Thank you for the info.  Do you have  a pointer for the Milight Tweakers Forum? The only forum I am aware of is the Dutch Tweakers.net forum at <https://tweakers.net/merk/3235/milight/forum/>. If this is the forum you're refering to, maybe you can point me to the interesting threads?! This would be great.

Thank you very much also for your offer to test stuff! I certainly will provide a test branch at the early stages of development.

rspaargaren, 2016-12-01:
@mwittig The forum I was refering to is https://gathering.tweakers.net/forum/list_messages/1650755/63
In the code provided by limitedless a wifibridge ID is mentioned. This is refering to a session id wich can be asked by sending out the following sequence: 
UDP 20 00 00 00 16 02 62 3A D5 ED A3 01 AE 08 2D 46 61 41 A7 F6 DC AF D3 E6 00 00 1E
The wifibridge will answer with a hex sequence as well. The 20th HEX code is the (temporaly) wifibridg ID. By using this ID I was able to switch on / of the bridge lamp as a test.
The info can be read on the last pages of the forum. (29th of November)

mwittig, 2016-12-02:
@RSpaargaren As far as I understand the conversation on Tweakers.net you have some basic prototyping code running for Milight v6? If yes, are you ready to share this with me? Maybe via gist? This would be great.

Update from my side: Thanks to @Oggymator I now have a v6 bridge to play with. I have started to implement the discovery part  (very easy)  and started to reverse engineer the wire-protocol.  Obviously, the  API overview given on the web site is obviously incomplete and faulty. I am wondering why these guys call it an "Opensource API"! It is perhaps a cultural misunderstanding on my side, but I think it should be better called the "API Challenge for Open Source Developers".  Well, no need to argue on this. Most developers like these challenges as solving them is demonstrating how smart they are ;)

rspaargaren, 2016-12-02:
Hi, unfortunately I don't have prototyping software running. What I did was a manual interaction with the wifi bridge via Packet sender. Basically I have followed the following protocol.
Send out the following HEX UDP  package to wifi bridge port 5987
UDP 20 00 00 00 16 02 62 3A D5 ED A3 01 AE 08 2D 46 61 41 A7 F6 DC AF D3 E6 00 00 1E
The Wifi Bridge responds with a hex string and I pick the 20 th HEX code as wifibrdige ID.
_◾UDP.SEND hex bytes: 20 00 00 00 16 02 62 3A D5 ED A301 AE 08 2D 46 61 41 A7 F6 DC AF D3 E6 00 00 1E 
◾UDP.Response: 28 00 00 00 11 00 02 (AC CF 23 F5 7A D4)(mac) 69 F0 3C 23 00 01 **05** 00 00_ 
Then I have used this value in the following example string to turn the lamp wifibridge on / off
WB = for example 05
SN = 01
_◾LimitlessLED Wifi Bridge Light ON 80 00 00 00 11 WB 00 00 SN 00 (31 00 00 00 03 03 00 00 00)(cmd) 01(zone) 00 38(chksum) UDP response: (88 00 00 00 03 00 SN 00) 
◾LimitlessLED Wifi Bridge Light OFF 80 00 00 00 11 WB 00 00 SN 00 (31 00 00 00 03 04 00 00 00)(cmd) 01(zone) 00 39(chksum) UDP response: (88 00 00 00 03 00 SN 00)_ 

This has worked in practice but it has now to be interpretated ito workable code.

Oggymator, 2016-12-02:
@mwittig 
The most people dont know it, but LimitlessLED is not the producer or inventor of these lights and products.
At the moment he is only a reseller like me. In the past he was involved on the first wifi bridge, if my informations are correct. The real company behind the products came from china and have also some more porducts like LimitlessLED and me.

mwittig, 2016-12-02:
@Oggymator thank you very much for the clarification. This seems also to be the case for Milight.com as the "About Us" page prompts the folllowing: "Gigateq Limited is an electronic retailer based in Manchester, United Kingdom. We are the largest independent online manufacturer and retailer of the innovative easybulb LED bulb that can be controlled with smartphones and mobile devices".

mwittig, 2016-12-02:
@rspaargaren Thank you very much for getting back to me. I'll do my best develop a first draft based on the available information.

Oggymator, 2016-12-02:
@mwittig
Yes, MiLight.com is also only a reseller from the Chinese company with its brand Easybulb.
They was fast enough for the domain milight.com ;)
We have the domain mi-light.de so it is not easy to find the real Producer with the domain.
It is also not easy to communicate with the real Chinese company about the API, but I can try if it will help you.
For the last times I got the LimitlessLED API information and never talked with the Chinese company about that.
Regards Oggy

rspaargaren, 2016-12-02:
@mwittig @Oggymator how About futlight.com? Are they the real supplier? They seam to have a complete and new range on their website?

Oggymator, 2016-12-02:
You found it ;)

duecedriver, 2016-12-02:
found the api for the milight v3 / limitless v6 on the limitless site..

Section 1. Manual Web Browser Wifi Bridge v6 Setup.
Connect Phone to Milight AccessPoint
Use Phone Web Browser to http://10.10.100.254
Username: admin
Password: admin
Work Mode: change to STA mode, press Save
STA Setting: Scan for your Home Wifi Router
Encryption Method: WPA2PSK
Encryption Algorithm: AES
Password: enter your home wifi router password here.
Obtain an IP address automatically: Enable
Restart: Click Ok
Now put your phone back on your home wifi network, and open the app, it should see the wifi bridge, assuming you entered your wifi password correctly a few steps back. :)
Section 2. Searching for LimitlessLED v6 Bridge/s on the LAN.
We have a Windows App for this called LimitlessLED Admin tool v5/v6
If you are building your own Program, use the following (If you want more exact C# code for this, email us)
UDP.IP = “255.255.255.255″; // IP.Broadcast
UDP.PORT = 48899;
UDP.SendBytes(UTF8.GetBytes(“HF-A11ASSISTHREAD”)); // v6 Bridge (for v5 use “Link_Wi-Fi”)
receiveBytes = UDP.Receive();
response = UTF8.GetString(receiveBytes);
//returns IP address of the wifi bridge, the unique MAC address, and the name(which is always the same)
// 10.1.1.27,ACCF23F57AD4,HF-LPB100
// 10.1.1.31,ACCF23F57D80,HF-LPB100
Section 3. Wifi Bridge v6 Protocol
UDP.IP = “255.255.255.255″; (or direct Wifi Bridge IP Address)
UDP.PORT = 5987;
UDP.SEND (hex bytes, see below)
SN = Sequential Hex Number (Sequence Number) 0×01 -> 0xFF
WB = Wifi Bridge ID 0×01 -> 0xFF
RGBW/WW/CW Commands
UDP Hex Send Format: 80 00 00 00 11 {WifiBridgeID} 00 00 {SequenceNumber} 00 {COMMAND} {ZONE NUMBER} 00 {Checksum}
UDP Hex Response: 88 00 00 00 03 00 {SequenceNumber} 00
List of {COMMAND}s:
31 00 00 08 04 01 00 00 00 = Light ON
31 00 00 08 04 02 00 00 00 = Light OFF
31 00 00 08 04 05 00 00 00 = Night Light ON
31 00 00 08 05 64 00 00 00 = Dual White Light ON (Color RGB OFF)
31 00 00 08 01 BA BA BA BA = Set Color to Blue (BA) (FF = Red, D9 = Lavender, BA = Blue, 85 = Aqua, 7A = Green, 54 = Lime, 3B = Yellow, 1E = Orange)
31 00 00 08 02 SS 00 00 00 = Saturation (SS hex values 0×00 to 0×64 : examples: 00 = 0%, 19 = 25%, 32 = 50%, 4B, = 75%, 64 = 100%)
31 00 00 08 03 BN 00 00 00 = BrightNess (BN hex values 0×00 to 0×64 : examples: 00 = 0%, 19 = 25%, 32 = 50%, 4B, = 75%, 64 = 100%)
31 00 00 08 05 KV 00 00 00 = Kelvin (KV hex values 0×00 to 0×64 : examples: 00 = 2700K (Warm White), 19 = 3650K, 32 = 4600K, 4B, = 5550K, 64 = 6500K (Cool White))
31 00 00 08 06 MO 00 00 00 = Mode Number (MO hex values 0×01 to 0×09 : examples: 01 = Mode1, 02 = Mode2, 03 = Mode3 .. 09 = Mode9)
31 00 00 08 04 04 00 00 00 = Mode Speed Decrease–
31 00 00 08 04 03 00 00 00 = Mode Speed Increase++
3D 00 00 08 00 00 00 00 00 = Link (Sync Bulb within 3 seconds of lightbulb socket power on)
3E 00 00 08 00 00 00 00 00 = UnLink (Clear Bulb within 3 seconds of lightbulb socket power on)
Wifi Bridge iBox LED Lamp {COMMAND}s (Zone Number = 0×01)
31 00 00 00 03 03 00 00 00 = Wifi Bridge Lamp ON
31 00 00 00 03 04 00 00 00 = Wifi Bridge Lamp OFF
31 00 00 00 04 MO 00 00 00 = Mode Number (MO hex values 0×01 to 0×09 : examples: 01 = Mode1, 02 = Mode2, 03 = Mode3 .. 09 = Mode9)
31 00 00 00 03 01 00 00 00 = Mode Speed Decrease–
31 00 00 00 03 02 00 00 00 = Mode Speed Increase++
31 00 00 00 01 BA BA BA BA = Set Color to Blue (BA) (FF = Red, D9 = Lavender, BA = Blue, 85 = Aqua, 7A = Green, 54 = Lime, 3B = Yellow, 1E = Orange)
31 00 00 00 03 04 00 00 00 = Set Color to White (is ignored when Lamp is OFF, it does NOT turn the Lamp ON)
31 00 00 00 02 BN 00 00 00 = BrightNess (BN hex values 0×00 to 0×64 : examples: 00 = 0%, 19 = 25%, 32 = 50%, 4B, = 75%, 64 = 100%)
Valid List for {ZONE NUMBER}
0×00 All
0×01 Zone1
0×02 Zone2
0×03 Zone3
0×04 Zone4
Checksum:
RGBW/WW/CW Checksum Byte Calculation is the sum of the last 11 bytes before end of the UDP packet. The checksum is then added to the end of the UDP message.
e.g. SUM((31 00 00 08 04 01 00 00 00)(command) 01(zone) 00) = 3F(chksum)
EXAMPLES
LimitlessLED Wifi Bridge Light ON 80 00 00 00 11 WB 00 00 SN 00 (31 00 00 00 03 03 00 00 00)(cmd) 01(zone) 00 38(chksum) UDP response: (88 00 00 00 03 00 SN 00)
LimitlessLED Wifi Bridge Light OFF 80 00 00 00 11 WB 00 00 SN 00 (31 00 00 00 03 04 00 00 00)(cmd) 01(zone) 00 39(chksum) UDP response: (88 00 00 00 03 00 SN 00)
[... 33 lines cut ...]
example3: 10 00 00 00 0A 02 FE 51 01 (AC CF 23 F5 7D 80)(MAC address)
Section 4. Smart Link Setup Services
** Warning only have one wifi bridge powered up at a time when setup using smart link (In case you have more than one in your home) **
Smart-link is all about getting your brand new wifi bridge onto your home wifi router network.
SmartLink saves the user time during the setup phase, from having to switch to the bridge in the wifi settings to set it up. Instead it is done using a sync button underneath the Wifi Bridge v6 and an app.
If you want to build smart-link into your own app.
SMART LINK UDP port 49999 Default Password “8888″
UDP.IP = “10.1.1.255″
UDP.PORT = 49999
UDP.Send Whilst Waiting: 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05 05
Section 5. Cloud Remote Internet access Services
If you want to build an app that controls the lights over the internet, more information will be made available here.
http://www.xlink.cn/
[... cut ...]
Default Cloud Port: 23778
Cloud Server Login TCP.Send: 10 00 00 00 1A 02 4B 4B 59 DB 00 10 32 32 30 66 61 32 61 66 66 65 36 63 61 32 30 30 00 00 3C
Cloud TCP.Read Login Response: 18 00 00 00 02 00 00
Cloud Keep Alive: TCP.Send: D0 00 00 00 00
Cloud Keep Alive: TCP.Read: D8 00 00 00 00
Cloud TCP sent: A0 00 00 00 07 69 F0 3D F3 00 0B 00
Cloud response: A8 00 00 00 13 69 F0 3D F3 00 0B 00 07 00 09 x l i n k _ d e v
Cloud TCP sent:
Cloud response: 6{“device_id”:1777352179,”type”:”offline”,”operator”:0}

mwittig, 2016-12-02:
@duecedriver Thank you very much. I have edited your posted and replaced the content with the link to the originating site.

duecedriver, 2016-12-02:
thanks... I use home assistant and doesnt look like the new v3/v6 hub is supported there yet either...

hope this api info can help you guys get this working !!!

duecedriver, 2016-12-04:
Hamish... at limitlessled in NZ and i have been talking via email and he is working with developers and writing example scripts for them.. he is a development partner with the mfg and has the source code for the v6 bridge... contact him at limitlessled and I am sure he will give you anything you need... no need for reverse engineering if he can give you the whole book chapter and verse...

Hoping that you and the home assistant guys cant get the code up and running on this. Would love to add this bridge to my setup

rspaargaren, 2016-12-04:
He is quite active on the following NZ forum:
http://m.geekzone.co.nz/Forums/73/Topic/113569/50

mwittig, 2016-12-09:
Brief update: I queried LimitlessLED for more technical information and received a friendly response. Following this, the website had been updated with some more details. It was also promise some example code will be published soon.

Based on the information provided by @rspaargaren I have some basic code up running which switches the bridge LED, extracts the WifiBridgeSession bytes, builds the message buffer from the various parameters,  and does the checksum calculation. However, the code messy and it will take some more days to clean-up and improve it for a first drop.

rspaargaren, 2016-12-09:
Hi @mwittig good work!, I am looking forward for your first beta version! I am also happy to read that v7 of the software is also comming where we can control individual lights etc. I expect the change in code in v6 is already preparing for that...

rspaargaren, 2016-12-12:
Hi @mwittig did you check the limitless website? They have now added a nodejs example code ! Or did you create this code..?

mwittig, 2016-12-12:
No, I didn't create this code. The example is incomplete unfortunately, but still it is very helpful

rspaargaren, 2016-12-14:
Here another piece of code....

https://github.com/daniel-t/MiLight_v6

mwittig, 2016-12-15:
@rspaargaren That's great. The code published on the Limitless LED web seite for nodeJS appears to be excerpt of the code included in this project!

I am making good progress and have working code for most of the command set including WW/CW white bulb commands which are not including in the info published so far.

mwittig, 2016-12-21:
Sorry for the delay at my end, as other things kept me busy most of the time. I have now pushed a first batch of the V6 implementation to the repo as promised. 

I am planning to release the code before end of the year. So, if you guys have some time to play with the code and provide feedback this will be great! Just have a look at the examples to start with.

rspaargaren, 2016-12-23:
great also here a bit busy, but I will have a look! Keep up the good work.

mwittig, 2017-01-24:
Released version 0.1.0

I am closing this now. Please open a new ticktet for any related issues or questions. Thanks again for your support in getting V6 support realized as part of node-milight-promise! Certainly, I'll continue improving the  library.

closed 2017-01-24 by mwittig