mwittig/node-milight-promise #13: How to control RGB+CCT color temperature in white mode?

odelma, 2017-01-02:
Hi, thanks for this great library. At last I can integrate some functionality also to my home automation system (Domoticz)! 😄

Unfortunately I couldn't figure out if there is a way to control the color temperature of white mode (warm / cold) with "FullColor" bulbs? Seems that when "whitemode" is set, the temp goes to coldest. Is there a way to set a warmer color temp for white?

Once again, thanks a lot for your great work!

mwittig, 2017-01-02:
Hi, thank your very much for using node-milight-promise and the positive feedback! 

Actually, changing the the white color temperature is currently missing, but I'll add this asap.

odelma, 2017-01-03:
Thanks a lot! Now I understand why I was not able to figure it out earlier :smile: Will test later this evening...

closed 2017-01-04 by mwittig