mwittig/node-milight-promise #14: When you plan to publish your new version on npm?

GermanBluefox, 2017-01-15:
Because of V6 support?

GermanBluefox, 2017-01-15:
Btw, this code is working for me: https://github.com/Daniel-t/MiLight_v6/blob/master/milight.js

mwittig, 2017-01-15:
Hi, thank you very much for looking into node-milight-promise! 

The only thing missing right now is session management which I am currently working on. The  code is on master apart from the session management code which I am currently developing. See also #7 for details. 

I am aware of the other project which has some shortcomings though. My intent is to support the v6 API and the legacy API in a single driver and to provide some glue code to simplify the integration. Note, the legacy and the v6 API use different color wheels for the "old" RGBWW bulbs, for example. See README for details.

I am shooting for a new release by end of the month. If you like the project please consider "starring" the project. Thanks!

closed 2017-01-15 by mwittig