mwittig/node-milight-promise #15: Add support for keep-alive command messages

mwittig, 2017-01-17:
This requires a better understanding of what can be done with keep-alive messages.  The behavior of keep-alives as part of the Milight protocols isn't documented.

AniTigran, 2017-03-26:
sombody can give me milightv 6 modul end send comand

mwittig, 2017-03-26:
@AniTigran Can you please give some more details about what you need? Please also have a look at the [examples](https://github.com/mwittig/node-milight-promise/tree/master/examples). Generally, the flow is as follows:

```
var light = new Milight({
    ip: "192.168.0.34",
    type: 'v6'
  }),
  zone = 1;

// Send commands
light.sendCommands(commands.rgbw.on(zone));

// ...

// Finall close the connection to terminate the application
// It will close when all pending commands have been processed
light.close().then(function () {
  console.log("All command have been executed - closing Milight");
});

AniTigran, 2017-03-29:
I have asmart home in majordomo,but I can not connect to the smart home milightV6 it dose not recognized.Ithink that the modul of wifi Ibox and send comande not work can you help me to connect 

>
>@AniTigran Can you please give some more details about what you need? Please also have a look at the  examples . Generally, the flow is as follows:
>var light = new Milight({
    ip: "192.168.0.34",
    type: 'v6'
  }),
  zone = 1;

// Send commands
light.sendCommands(commands.rgbw.on(zone));

// ...

// Finall close the connection to terminate the application
// It will close when all pending commands have been processed
light.close().then(function () {
  console.log("All command have been executed - closing Milight");
});

mwittig, 2017-03-29:
Did you try "discover" your Ibox? Try to run "discover-bridges-v6.js" located in the examples directory

AniTigran, 2017-04-01:
I can not connect milightv6 to my smart home it not respond. I put the IP of milightv6,numbers of zones,and color of the led,make a send comande on but there no aswer from milight.in configration of old milight in my smart home I must put a milght modul php ,milght class,and send comand in all.My smart  home platform is MAjordomo.

mwittig, 2017-04-02:
Did you pair your lights with the ibox controller?

AniTigran, 2017-04-02:
Yes I do it work corectly by the progrom OF LIMITLESSLED .Led lights work but not work with smart home

mwittig, 2017-04-02:
Did you try the example programs as sugested earlier? Are you sure majordomo is using node-milight-promise? As far as I can see majordomo is written in PHP and has its own set of PHP plugins like https://github.com/Gelezako/MajorDomo-MiLight

AniTigran, 2017-04-02:
Yes of cours Itry but without result.what about  milight v6 PHP plgins I finde  https://github.com/Uhula/IPSMilightV6  in IPS Milightv6 I finde some module milghtV6 php  ,but with this module it not respond it like not understend comands.is it try?and what I must finde new send comand?

mwittig, 2017-04-04:
Sorry, I am a bit lost here. As far as I can see "IPSMilightV6" and "MajorDomo-MiLight" are completely unrelated to my project. Maybe you can use "IPSMilightV6" as a basis for a MajrorDomo Milight V6 plugin. 

If you want to trial my project  please try to execute the examples as suggested earlier.

markg85, 2017-12-16:
You can find an 'exmaple' implementation of keep-alive here as well: https://github.com/JasperG/bilight/blob/master/bilight/src/main/java/com/jaspergoes/bilight/milight/Controller.java#L774

That is from the 'bilight' app