mwittig/node-milight-promise #16: How to send commands to wifi box?

webfrank, 2017-02-13:
I have a V6 WiFi iBox with light and no other bulbs at the moment, but I cannot find a way to send commands to the bridge itself.

mwittig, 2017-02-13:
See https://github.com/mwittig/node-milight-promise/blob/master/examples/example-bridge-v6.js which is an example on how to control the bridge light. Instead of providing the global broadcast address which is the default, you should provide the IP address of the Milight Controller for unicast mode. Don't use the global broadcast address on Windows as this may give unexpected results.

mwittig, 2017-02-13:
I think the question has been answered. Please re-open if you want to follow-up on this

closed 2017-02-13 by mwittig