mwittig/node-milight-promise #17: Script hangs?

ygalanter, 2017-02-23:
Sorry for probably really stupid question, I tried to create basic script based on your example, but it hangs on object creation. If I leave this part in the code (192.168.1.24 is IP of my bridge):
```
var light = new Milight({
        ip: "192.168.1.24",
        delayBetweenCommands: 75,
        commandRepeat: 2
 });
```
it simple hangs, no errors, but it doesn't proceed any further. I am probably missing something basic, but any idea what it can be?

mwittig, 2017-02-23:
Hi, thanks for trying node-milight-promise!

Your script hangs as the `light` object has opened event listeners for the UDP connection to your milight bridge. That's common node.js behaviour, btw. - a node.js app won't terminate if there event listeners waiting for events.

See [example-rgbw.js](https://github.com/mwittig/node-milight-promise/blob/master/examples/example-rgbw.js) where you can see commands being send to the bridge like `light.sendCommands(commands.rgbw.on(zone))`. Finally, at the end of the program you have light.close(). This will close all event listeners when all pending commands have been sent to bridge. So, if you put `light.close()`at the end of you code it should terminate as expected. Hope this helps

ygalanter, 2017-02-23:
Thank you very much for the prompt response!

Yes this module is godsent for v6 milight. Now instead of sending raw multibyte codes I can work with cool objects and methods.

Oh, I thought all commands are sequential – object is created and then commands follow. Am I understanding correctly that following commands will still execute even though listener is waiting?

Thanks!

Yuriy.

From: Marcus Wittig 
Sent: Thursday, February 23, 2017 12:21 PM
To: mwittig/node-milight-promise 
Cc: ygalanter ; Author 
Subject: Re: [mwittig/node-milight-promise] Script hangs? (#17)

Hi, thanks for trying node-milight-promise!

Your script hangs as the light object has opened event listeners for the UDP connection to your milight bridge. That's common node.js behaviour, btw. - a node.js app won't terminate if there event listeners waiting for events.

See example-rgbw.js where you can see commands being send to the bridge like light.sendCommands(commands.rgbw.on(zone)). Finally, at the end of the program you have light.close(). This will close all event listeners when all pending commands have been sent to bridge. So, if you put light.close()at the end of you code it should terminate as expected. Hope this helps

mwittig, 2017-02-23:
Yuriy, actually commands are added to a promise chain. Each command returns a promise, and the next command will only be executed if the previous promise has been settled. The same is true for close() . The actual tear down of the UDP socket and removal of the even listeners is performed no earlier than the last command promise has been settled.

If you like node-milight-promise please consider starring the project. Thanks

mwittig, 2017-02-23:
If I am getting you right you have one of these new bridges which use the new "v6" protocol. If this is the case the setup of the Milight object needs to be slightly different: 

```
var light = new Milight({
        ip: "192.168.1.24",
        type: "v6"
 });
```
You don't need to set delayBetweenCommands and commandRepeat for v6 as the new flow control feature will handle that for you.  Please have look at the README and the examples for more details.

ygalanter, 2017-02-23:
You’re right, got v6 bridge, thanks for the pointers. Starred the project because it’s awesome.

Also this is what the doctor ordered for Alexa/Google Assistant voice integration. HA-Bridge (https://github.com/bwssytems/ha-bridge)  allows to run custom scripts, so I plan to call code using your module from ha-bridge. 

Thanks,

Yuriy.

From: Marcus Wittig 
Sent: Thursday, February 23, 2017 3:16 PM
To: mwittig/node-milight-promise 
Cc: ygalanter ; Author 
Subject: Re: [mwittig/node-milight-promise] Script hangs? (#17)

If I am getting you right you have one of these new bridges which use the new "v6" protocol. If this is the case the setup of the Milight object needs to be slightly different:

var light = new Milight({
        ip: "192.168.1.24",
        type: "v6"
 });
You don't need to set delayBetweenCommands and commandRepeat for v6 as the new flow control feature will handle that for you. Please have look at the README and the examples for more details.

mwittig, 2017-02-24:
Yuriy, thanks a lot! I am closing this issue now. 
Please, feel free to open a new issue if you have further questions

closed 2017-02-24 by mwittig

ygalanter, 2017-02-25:
Thanks again for your help, it's working beautifully. I've blogged about how it helps with Amazon Echo voice commands, perhaps might be useful as usecase showcase: http://codecorner.galanter.net/2017/02/24/full-control-of-your-limitless-ledmilight-v6-bulbs-from-amazon-echo/
Thanks!

mwittig, 2017-02-25:
Yuriy, excellent! Thanks for the pointer. I'll definitely add a link to your article to the README with the next release as I think it is very useful for other users.