mwittig/node-milight-promise #18: Inconsistency in rgbw and full-color

greggilbert, 2017-03-25:
Hey there. First off, awesome library - gamechanger for me.

I've got a v6 bridge ([this one](https://ae01.alicdn.com/kf/HTB1ZzeZOXXXXXcuapXXq6xXFXXXk/Mi-Light-Wifi-ibox-2-4G-Wireless-LED-RF-Dimmable-Remote-By-APP-iOS-Android-for.jpg_640x640.jpg)), and I have one [RGBW/WW/CW Full Color](http://www.limitlessled.com/wp-content/uploads/2016/01/RGBWWCW1.png) and one [RGBW Color](http://www.limitlessled.com/wp-content/uploads/2012/11/SingleLight03.jpg) bulb attached to it. When I run the same rgb commands for each, I get different results.

```
var light = new Milight({
    ip: "10.0.0.20",
    type: 'v6'
  }),
  zone = 1;

// this turns the bulb blue
light.sendCommands(commands.fullColor.on(zone), commands.fullColor.whiteMode(zone), commands.fullColor.brightness(zone, 100));
light.pause(1000);
light.sendCommands(commands.fullColor.rgb(zone, 200, 100, 100));

// this turns the light red
light.sendCommands(commands.rgbw.on(zone), commands.rgbw.whiteMode(zone), commands.rgbw.brightness(zone, 100));
light.pause(1000);
light.sendCommands(commands.rgbw.rgb(zone, 200, 100, 100));
```

It's possible I'm missing something here, but I thought I'd ask. Any ideas?

mwittig, 2017-04-03:
Sorry for the delay in reply. I'll look into this asap.

mwittig, 2017-04-04:
Greg, thanks for reporting and the positive feedback. I'm able to reproduce the issue and it should be fixed shortly. Please note the color rendition (even for "full color") is limited: 255 hue values instead of 360 degrees hue, 20 native brightness levels instead of 100 eventhough the API is suggesting 100 levels, quality of the saturation is unknown. You'll notice the full color light for for #C86464 is much lighter than what the color table suggests.

mwittig, 2017-04-04:
published node-milight-promise@0.1.2

closed 2017-04-04 by mwittig

greggilbert, 2017-04-04:
Thanks! Just to be clear, the color rendering limitation is a Milight hardware issue, right?

mwittig, 2017-04-04:
Yes, that's right!