mwittig/node-milight-promise #21: Programmatic light bulb syncing

bartbutenaers, 2017-06-06:
Dear Marcus,

I'm a newbie in the Milight world, and I'm wondering if this node can be used to programmatically link my Milight light bulbs to a specified zone?

**Current setup without NodeJs**

`Milight App --> V6 Wifi bridge (iBox 2) --> light bulb`

When powering up a light bulb, you have 5 seconds in the Milight App to press a zone number X (from 1 to 4).  From that moment on, the light bulb is linked to zone number X.  

Remark: when another app is used (e.g. on another smartphone), another zone can be setup for the same light bulb.

**New setup with NodeJs**

`Web application --> Node-Milight-Promise --> V6 Wifi bridge (iBox 2) --> light bulb`

I would like to simulate the same behavious, without having to use my original Milight app (or remote control).  So when I press some button in my own web application (e.g. Node-Red), I want to let the Node-Milight-Promise node know that the light bulb (that is currently starting up) needs to be linked to zone X.

It is not clear to me if this is currently possible with this node.  If not, is this perhaps somewhere on the roadmap?

Thanks in advance !!!!!
Bart Butenaers

mwittig, 2017-06-09:
Hi Bart,

Thanks for using node-milight-promise! Do you have written your own Node-Red integration? In case, I am interested in learning more about your project if you're happy to share it.

> It is not clear to me if this is currently possible with this node. If not, is this perhaps somewhere on the roadmap?

When the bulb has not been linked yet to the controller it is good enough to send the .on() command a few times. This will link the bulb to the given zone of the iBox. According to my findings a bulb can only be assigned to a single zone of the iBox. If you wish to change the zone, you need to unlink the bulb from the given zone first. To fully support this, I have add the link/unlink V6 commands for all bulb types supported. It may take a few days to get this released, as I need to add tests and documentation for the new feature.

If you like this plugin, please consider &#x2605; starring [the project on github](https://github.com/mwittig/node-milight-promise). Thanks!

Marcus

mwittig, 2017-06-10:
Published node-milight-promise@0.1.4

closed 2017-06-10 by mwittig

bartbutenaers, 2017-06-13:
Hey Marcus,

we are currently experimenting with an existing [node-red-contrib-milight](https://www.npmjs.com/package/node-red-contrib-milight) node.  There is also another node-red-contrib-milight-wrapper node available (which adds some extra commands), but we haven't tried that yet.  Now we will have to experiment with these linking and unlinking options in Node-Red ...

Thanks for you fast replay!  You deserved your Github star ;-)

Kind regards,
Bart

mwittig, 2017-06-14:
Hi Bart, 

thanks for starring the project and the info on your work with node-red. I am glad you like the project. 

I briefly looked into node-red-contrib-milight which does not seem to support the new "ibox" and "ibox2" wifi controllers! Is that something you're tyring to do? If you need assistance in extending the node-red-contrib-milight code please let me know. It should be relatively straightforward to do.

KR, Marcus

bartbutenaers, 2017-06-17:
Hi Marcus,

I don't own an iBox2 myself, but a collegue at work is struggling to get it working via Node-Red.  So he asked me to get into contact with you.    But indeed, he is trying to connect it's iBox2 wifi controller using that node.  We planned to dig into the node-red-contrib-milight code in a couple of weeks.  I don't want to create a new Node-Red node, since Nick O'Leary already [asked](https://groups.google.com/forum/#!topic/node-red/6kMPCretQvg) the developer of the node-red-contrib-milight-wrapper node to merge both nodes instead of creating new ones ...

But if you already can give some tips or advices, you are very welcome!  E.g. you say:
`which does not seem to support the new "ibox" and "ibox2" wifi controllers`
Do we need to add some extra functionality to get around this?

Thanks for your time!
Bart

HellandB, 2017-07-18:
Hi Bart,

Did your friend figure out how to connect his ibox2/Limitless bridge v6 to Nodered?
I am in the same position having an unused v6 here that I can't integrate into my home automation system. I can't find any information on internet on how to set up the v6 for node-red. Would love if this node-milight-promise node got upgrade with v6 support. V5 version of bridge working flawlessly with the node.

Thanks.

mwittig, 2017-07-18:
@bartbutenaers, @HellandB, et al.

I reviewed the project for node-red-contrib-milight (https://github.com/stephenkeep/node-red-contrib-milight) and spent some time to add support for the "full color" bulb and bridge light. The code currently resides in my fork as I want to do more testing. If you are impatient and want to give it a try you should be able to install from my fork if you have git installed: `npm i mwittig/node-red-contrib-milight`

HellandB, 2017-07-19:
Wow! This great news. Thank you so much for looking into this. I'm busy the next weeks, but hope to have some time to play with it. Ill see if I try your fork or if its already merged into the master project.

bartbutenaers, 2017-07-19:
Hi Marcus,

my friend is in holiday currently.  Once he is back, he will test it.  Thanks very much for analyzing and contributing the node-red node.  Great support !!!!

Kind regards,
Bart

mwittig, 2017-07-19:
Bart,

you're welcome! As another improvement I have implemented a new "rgb" command which is useful if you want to use MiLight along with dashboard, for example. This way, it is possible to use the color picker. I am sure it is also useful in other scenarios. Meanwhile, I have a filed a pull request to the project. 

Kind Regards,
Marcus

HellandB, 2017-07-25:
Just tried your fork, installing just fine, but I get a error message when trying to push commands. I copied the example flow from the readme, and if I try push on (or any other button) node-red throws: 
"Milight error: RangeError: Offset into buffer too large"

mwittig, 2017-07-27:
Hi. The example flow may be outdated. I need to test it myself.

mwittig, 2017-07-27:
Just tried the example flow which basically works for me. Which node-red version do you have? Please make sure you have node-red 1.7.x installed, Please also let me know which nodejs version do you use.

I have slightly updated the example flow:
```
[{"id":"b5ab74d6.08edc8","type":"MiLight","z":"44cc4fb7.965b3","name":"White Bulb / Legacy Bridge","bridgetype":"legacy","bulbtype":"white","zone":1,"ip":"255.255.255.255","broadcast":true,"x":1035.765625,"y":297,"wires":[]},{"id":"51aa22c0.74d05c","type":"MiLight","z":"44cc4fb7.965b3","name":"Color Bulb / Legacy Bridge","bridgetype":"legacy","bulbtype":"rgbw","zone":1,"ip":"255.255.255.255","broadcast":true,"x":1036.7656211853027,"y":619.0000314712524,"wires":[]},{"id":"26ce02ea.cd2e6e","type":"inject","z":"44cc4fb7.965b3","name":"Off","topic":"","payload":"off","payloadType":"str","repeat":"","crontab":"","once":false,"x":534.7656402587891,"y":553.0000228881836,"wires":[["51aa22c0.74d05c"]]},{"id":"17bef0b.fe9ce0f","type":"inject","z":"44cc4fb7.965b3","name":"On","topic":"","payload":"on","payloadType":"str","repeat":"","crontab":"","once":false,"x":534.7656555175781,"y":607.0000152587891,"wires":[["51aa22c0.74d05c"]]},{"id":"44e9d122.6a83b","type":"inject","z":"44cc4fb7.965b3","name":"white","topic":"","payload":"white","payloadType":"str","repeat":"","crontab":"","once":false,"x":536.765625,"y":662.9999933242798,"wires":[["51aa22c0.74d05c"]]},{"id":"b477791.b31d088","type":"inject","z":"44cc4fb7.965b3","name":"Brightness 10%","topic":"","payload":"10","payloadType":"num","repeat":"","crontab":"","once":false,"x":568.765625,"y":732,"wires":[["e01cd3d0.06b27"]]},{"id":"e01cd3d0.06b27","type":"function","z":"44cc4fb7.965b3","name":"","func":"msg.command = 'brightness';\nreturn msg;","outputs":1,"noerr":0,"x":741.7656478881836,"y":757.0000553131104,"wires":[["51aa22c0.74d05c"]]},{"id":"865a0a9b.be0848","type":"inject","z":"44cc4fb7.965b3","name":"Color","topic":"","payload":"20","payloadType":"num","repeat":"","crontab":"","once":false,"x":536.7656307220459,"y":843.0000419616699,"wires":[["f578b6a1.3d7948"]]},{"id":"f578b6a1.3d7948","type":"function","z":"44cc4fb7.965b3","name":"","func":"msg.command = 'color';\nreturn msg;","outputs":1,"noerr":0,"x":738.7657089233398,"y":845.000057220459,"wires":[["51aa22c0.74d05c"]]},{"id":"2cc2747e.452f4c","type":"inject","z":"44cc4fb7.965b3","name":"On","topic":"","payload":"on","payloadType":"str","repeat":"","crontab":"","once":false,"x":540.7656326293945,"y":168.00000381469727,"wires":[["b5ab74d6.08edc8"]]},{"id":"c8b4fa4d.e53078","type":"inject","z":"44cc4fb7.965b3","name":"Off","topic":"","payload":"off","payloadType":"str","repeat":"","crontab":"","once":false,"x":540.2656383514404,"y":215.00000190734863,"wires":[["b5ab74d6.08edc8"]]},{"id":"9fb07ff8.51bba","type":"inject","z":"44cc4fb7.965b3","name":"Warmer","topic":"","payload":"warmer","payloadType":"str","repeat":"","crontab":"","once":false,"x":538.7656211853027,"y":258.00000381469727,"wires":[["b5ab74d6.08edc8"]]},{"id":"38e7a6b0.c484ca","type":"inject","z":"44cc4fb7.965b3","name":"Cooler","topic":"","payload":"cooler","payloadType":"str","repeat":"","crontab":"","once":false,"x":542.2656402587891,"y":303.00001525878906,"wires":[["b5ab74d6.08edc8"]]},{"id":"287ae328.5c945c","type":"inject","z":"44cc4fb7.965b3","name":"Bright Up","topic":"","payload":"bright_up","payloadType":"str","repeat":"","crontab":"","once":false,"x":548.7656555175781,"y":346.00002670288086,"wires":[["b5ab74d6.08edc8"]]},{"id":"268eff9f.0e175","type":"inject","z":"44cc4fb7.965b3","name":"Bright Down","topic":"","payload":"bright_down","payloadType":"str","repeat":"","crontab":"","once":false,"x":557.7656326293945,"y":392.99999809265137,"wires":[["b5ab74d6.08edc8"]]},{"id":"7f19fef2.a20dc","type":"inject","z":"44cc4fb7.965b3","name":"Bright Max","topic":"","payload":"bright_max","payloadType":"str","repeat":"","crontab":"","once":false,"x":547.7656383514404,"y":438.0000286102295,"wires":[["b5ab74d6.08edc8"]]},{"id":"afe72b66.8856e8","type":"inject","z":"44cc4fb7.965b3","name":"Night","topic":"","payload":"night","payloadType":"str","repeat":"","crontab":"","once":false,"x":539.7656536102295,"y":485.00002098083496,"wires":[["b5ab74d6.08edc8"]]},{"id":"1c0123a9.9f827c","type":"inject","z":"44cc4fb7.965b3","name":"Brightness 100%","topic":"","payload":"100","payloadType":"num","repeat":"","crontab":"","once":false,"x":568.7656555175781,"y":778.750057220459,"wires":[["e01cd3d0.06b27"]]}]

rspaargaren, 2017-07-31:
Hi I have got a full color bulb and I am now testing with iphone homekit plugin as well for Node-red. Would it be possible to input the (HSB) Hue, Saturation, Brightness as color input to the milight?

mwittig, 2017-08-01:
@rspaargaren Not exactly. You can use RGB, HSL or CSS colors. Please find an example for HSL below.

```
[{"id":"1fcc95b.80dd16a","type":"MiLight","z":"9c9e3ed7.b6975","name":"MiLight","bridgetype":"v6","bulbtype":"fullColor","zone":"4","ip":"192.168.178.255","broadcast":false,"x":1484.765625,"y":620.75,"wires":[]},{"id":"da30ad47.5e377","type":"inject","z":"9c9e3ed7.b6975","name":"","topic":"rgb","payload":"hsl(0,100%,50%)","payloadType":"str","repeat":"","crontab":"","once":false,"x":1166.765625,"y":621.75,"wires":[["1fcc95b.80dd16a"]]}]

rspaargaren, 2017-08-01:
Hi thank for the example got the following function now to store the output and convert:
`if (msg.payload.hasOwnProperty("Saturation")) {
    flow.set("sat", Number(msg.payload.Saturation));
}
if (msg.payload.hasOwnProperty("Hue")) {
    flow.set("hue", Number(msg.payload.Hue));
}
if (msg.payload.hasOwnProperty("Brightness")) {
    flow.set("bri", Number(msg.payload.Brightness));
}

var Hue = flow.get("hue");
var Sat = flow.get("sat");
var Bri = flow.get("bri");

/* Calculates and stores the HSL components of this HSVColour so that they can
 * be returned be the getHSL function.
 */

  // determine the lightness in the range [0,100]
  var Lig = Math.round((2 - Sat / 100) * Bri / 2);

  // store the HSL components
  var Sat2 = Math.round(Sat * Bri / (Lig < 50 ? Lig * 2 : 200 - Lig * 2));
  
  Hue = Math.round(Hue);
  // correct a division-by-zero error
  if (isNaN(Sat2)) Sat2 = 0;

msg.payload = "hsl(" + Hue + "," + Sat2 + "%," + Bri + "%)";
//msg.payload = [Hue,Sat,Bri,0,Hue,Sat2,Lig];
msg.topic = "rgb";
return msg;`

mwittig, 2017-08-02:
Hi, @bartbutenaers 

It looks like your calc is discussed here <https://stackoverflow.com/questions/3423214/convert-hsb-hsv-color-to-hsl> amongst other approaches.  There is comment that the IsNan check is not required, but that's not true. So I think your code should work as expected. 

See also "jsBin Demo and the calc" which is the highest rated solution and provides a different approach.

as part of node-red-contrib-milight I am currently using the "color" package for color conversion  which has some limitations as the lacking hsv support. Therefore, I think I'll migrate to "tinycolor2" which better suit the needs. https://www.npmjs.com/package/tinycolor2

If you're arware of color conversion packages or alike please let me know.

rspaargaren, 2017-08-03:
Hi I have manually updated the plugin and I can confirm it is working very well! 
I must say the delay between my the input on my iPhone and the output at the lamp is considerable but that has nothing to do with the color.

mwittig, 2017-08-04:
Great! Regarding the delay I guess that is to homekit or the homekit plugin. The milight plugin should not add more than 200ms to that. You can always test that by adding a manual trigger node which send a commands string to the Milight node.

rspaargaren, 2017-08-04:
Part of the delay is caused that the output of the homekit app is quite severe. In the case that it has single messages for Hue, saturation and brightness. So in case you change you already have 2 x 200 ms. And if you move your finger that a number of changes so it gets in a que. Maybe the output of the hap plugin should be reviewed to optimize and get a single object.

I have also asked a pull request to get the color temperature working in the node-red plugin. https://github.com/stephenkeep/node-red-contrib-milight/pull/4
Unfortunately the homekit of Apple is still crap related to temperature and it sends out a hue and saturation value.

mwittig, 2017-08-04:
> And if you move your finger that a number of changes so it gets in a que. 

May be you mitigate that by debouncing calls as part of the flow chain. There is a "debounce" node which works well. I have been using that as part of my experiments. See https://flows.nodered.org/node/node-red-contrib-debounce

> Maybe the output of the hap plugin should be reviewed to optimize and get a single object.

Yes that would be nice. Maybe you can coin that into a feature request on the plugin project.

rspaargaren, 2017-08-04:
Hi I was already looking for such a feature in standard toolset but this looks like a good node!

Yes I will as for a feature request, or try to create a pull request...

bartbutenaers, 2017-09-05:
Marcus,

my collegue Robert just came by to tell me that he has tested your new version, and it _everything_ seems to be working fine now.   So thanks a lot for all the effort and free time you have spend to this issue !!

Kind regards,
Bart

mwittig, 2017-09-05:
Bart,

Sounds great! Thanks for the feedback. 

Unfortunately there is still no new npm package release even though a maintainer stepped in to help with the project. However, he has no access to npmjs and the author is not responding to e-mails. For this reason, I'll publish the plugin as a new project soon. I'll keep you posted on this.

Kind Regards
marcus