mwittig/node-milight-promise #22: Dual White v6 commands not working

RobinBol, 2017-06-19:
I have a Milight 6W DUAL WHITE LED Lamp E27 linked to a v6 bridge, but when using the white commands (for example the example-dual-white-v6.js script) the bulb does not react, when using the new mobile app (specifically the 10th remote in the remote selector), it does work. Could it be that the commands are not correct? Anyone else having issues with the dual white bulbs? RGBW is working fine by the way!

mwittig, 2017-06-19:
Works for me, but maybe I have an older version of the dual white bulb. Mine is working with controller 11 and 12 of the Milight iPhone app. 

[image]

[image]

mwittig, 2017-06-22:
@RobinBol any news on this? Maybe you can describe which bulb model you have and post a screenshot of the app control you're using. Thanks

RobinBol, 2017-06-22:
Sorry for my slow response, I was not able to test some things that I wanted to test before reporting back.

So it turns out, I paired the exact same bulb you show (according to the text on the bulb) but with the following remote:

[image]

So my guess was that this is causing the bulb to not work with your library. So I repaired the bulb with the remote you showed and it all worked (except for the off command, for which I found the correct one). But then to be sure I repaired the bulb again with the non-functioning remote; and again it worked. So I am not sure what the problem was, but it seems the commands are correct (except for the v6 -> white -> off).

Could you verify that with bridge v6 the off command for you white bulb works? I will create a PR for the off command that is working for me.

closed 2017-06-22 by RobinBol