mwittig/node-milight-promise #24: RGBW NightMode is a valid FullColor command

mpartoglou, 2017-06-24:
Hi,

Loving this - works a charm for me (using v6 bulbs). I have a mixture of RGBW and FullColor in my house - both support the nightmode feature. 

Just an option - you could include this in the commandsv6 so it can be run via: light.sendCommands(commands.fullcolor.nightMode(zone));

Alternatively, for others you can simply run it as below (same outcome).
light.sendCommands(commands.rgbw.nightMode(zone));

just an FYI and thanks again for creating this.

mwittig, 2017-06-25:
Hi @mpartoglou 

thanks for using node-milight-promise! The command `commands.fullcolor.nightMode(zone)` is already included and it should work. I have just re-tested it with my 8W RGB+CCT Milight bulb and it works for me (see sample below).

Marcus

```
var Milight = require('../src/index').MilightController;
var commands = require('../src/index').commandsV6;

// Important Notes:
// *  Instead of providing the global broadcast address which is the default, you should provide the IP address
//    of the Milight Controller for unicast mode. Don't use the global broadcast address on Windows as this may give
//    unexpected results. On Windows, global broadcast packets will only be routed via the first network adapter. If
//    you want to use a broadcast address though, use a network-specific address, e.g. for `192.168.0.1/24` use
//    `192.168.0.255`.

var light = new Milight({
    ip: "255.255.255.255",
    type: 'v6'
  }),
  zone = 0;

light.sendCommands(commands.fullColor.nightMode(zone));

light.close().then(function () {
  console.log("All command have been executed - closing Milight");
});
console.log("Invocation of asynchronous Milight commands done");
```

mpartoglou, 2017-06-25:
@mwittig thanks so much - apologies, seems to be working for me now. I will close this off!

closed 2017-06-25 by mpartoglou