mwittig/node-milight-promise #25: Enhancement: Support from v6 Bridge led

thofra, 2017-07-07:
The v6 bridge also provides a RGB led which can be controlled from the app. Any plans to include support for this bridge? Basically, this led always lights up when the bridge is powered up. I would like to switch it off in such cases w/o the need to open the app. Bridge led control is described in http://www.limitlessled.com/dev/ - thanks a lot for your efforts.

mwittig, 2017-07-07:
Hi. I am happy to do this. However, I don't think http://www.limitlessled.com/dev/  contains the v6 control codes for RGB as it doesn't for WW/CW. Please point me to the specific location if you have spotted them though.

Do you use the Smartphone App? If yes,  it'll be helpful to know which control panel you use to control your RGB lights! Is it possible for you to make a screenshot? In case you don't know how to do this I can advise further. Just let me know the type of Smartphone you have in case. 

As I don't RGB light to test with I'll need your help with testing stuff.

thofra, 2017-07-09:
Hi, it is a little bit hidden on that side. Please search for "Wifi Bridge" and you should find the following:

Wifi Bridge iBox LED Lamp {COMMAND}s (Zone Number = 0x01)
31 00 00 00 03 03 00 00 00 = Wifi Bridge Lamp ON
31 00 00 00 03 04 00 00 00 = Wifi Bridge Lamp OFF

31 00 00 00 01 BA BA BA BA = Set Color to Blue (BA) (FF = Red, D9 = Lavender, BA = Blue, 85 = Aqua, 7A = Green, 54 = Lime, 3B = Yellow, 1E = Orange)
31 00 00 00 03 05 00 00 00 = Set Color to White (is ignored when Lamp is OFF, it does NOT turn the Lamp ON)

31 00 00 00 02 BN 00 00 00 = BrightNess (BN hex values 0x00 to 0x64 : examples: 00 = 0%, 19 = 25%, 32 = 50%, 4B, = 75%, 64 = 100%)

Would be so great... thanks.

thofra, 2017-07-09:
[image]

mwittig, 2017-07-15:
Thanks for your feedback. I'll try to look into this asap

mwittig, 2017-07-27:
Sorry for the delay in reply.

> Hi, it is a little bit hidden on that side. Please search for "Wifi Bridge" and you should find the following - at least I think this is the right way to access it:
>Wifi Bridge iBox LED Lamp {COMMAND}s (Zone Number = 0x01)
31 00 00 00 03 03 00 00 00 = Wifi Bridge Lamp ON
31 00 00 00 03 04 00 00 00 = Wifi Bridge Lamp OFF

Please note, the commands and the screenshot of the control you've sent are for the iBox bridge light. This is already supported.  You can use it as follows:

```
var Milight = require('node-milight-promise');
var commands = Milight.commandsV6;

// Important Notes:
// *  Instead of providing the global broadcast address which is the default, you should provide the IP address
//    of the Milight Controller for unicast mode. Don't use the global broadcast address on Windows as this may give
//    unexpected results. On Windows, global broadcast packets will only be routed via the first network adapter. If
//    you want to use a broadcast address though, use a network-specific address, e.g. for `192.168.0.1/24` use
//    `192.168.0.255`.

var light = new Milight.MilightController({
    ip: "255.255.255.255",
    type: 'v6'
  });

light.ready().then(function() {
  // Switch On, White Mode, Brightness 100%
  light.sendCommands(commands.bridge.on(), commands.bridge.whiteMode(), commands.bridge.brightness(100));
  light.pause(1000);

  // Switch Off
  light.sendCommands(commands.bridge.off());
  light.pause(1000);

  // Switch On again
  light.sendCommands(commands.bridge.on());
  light.pause(1000);

  // Set Hue
  for (var x = 0; x < 256; x += 5) {
    light.sendCommands(commands.bridge.hue(x));
    if (x === 0) {
      light.sendCommands(commands.bridge.brightness(100))
    }
  }
  light.pause(1000);

  // Switch Off
  light.sendCommands(commands.bridge.off());
  light.pause(1000);

  // Back to white mode
  light.sendCommands(commands.bridge.on(), commands.bridge.whiteMode());
  light.pause(1000);

  // Setting Brightness
  light.sendCommands(commands.bridge.on());
  for (var x = 100; x >= 0; x -= 5) {
    light.sendCommands(commands.bridge.brightness(x));
  }
  light.pause(1000);

  // Switch Off
  light.sendCommands(commands.bridge.off());
  light.pause(1000);

  light.close().then(function () {
    console.log("All command have been executed - closing Milight");
  });
  console.log("Invocation of asynchronous Milight commands done");
}).catch(function(error) {
  console.log(error);
});;

closed 2017-07-27 by mwittig