mwittig/node-milight-promise #28: Getting the current state of a lightbulb?

msunjic98, 2017-09-08:
Hello. I'm wondering is it possible to get the current state of individiual zone or lightbulb (like is it on, off, what color it has at the moment etc.) ?

mwittig, 2017-09-08:
Hi, thanks for using node-milight-promise. Unfortunately, the Milight protocols (v4 & v6) do not provide for querying the bulb state. Hopefully, this will be provided with a future version of the Milight protocol. The solution for Apps is to either maintain the last known state (i.e. the last switch command executed by the app) or to design the App "stateless" (i.e. having buttons for "on" and "off"). The latter is the case for the Milight Apps for iPhone and Android.

msunjic98, 2017-09-08:
Can I rely on last command state? How reliable is the current protocol, any hiccups?

mwittig, 2017-09-09:
> Can I rely on last command state

Generally, it is not reliable. You always need to account that the light may have been swicthed by another control device or by a physical light switch (which will turn the bulb to on, always). The Milight protocols prior to v6 are not reliable and lossy. v6 is a bit better as it wifi controller will confirm received commands, however, I believe there is no confirmation for link between the wifi controller and the bulb. Normally this link works flawlessly unless the bulb is located in a lmap with metal body or the bulb is located far away from the controller. Unfortunately, it is not possible to obtain data on the signal quality of the bulb RF link from the wifi controller (I think it is because the link protocol is uni-directional). For the WiFi-part it is possible to obtain singnal strength data, however.   

> How reliable is the current protocol, any hiccups?

Taking into account my previous account, I'd say the protcol is reliable for the IP-part and the RF-link to the bulb(s) is good, the overall reliability is very good.

msunjic98, 2017-09-11:
I'll just send default state commands on start, and then rely on last state.

mwittig, 2017-09-11:
OK, sound reasonable to me.

I am closing this ticket now. Please re-open if you wish to follow up on the matter

closed 2017-09-11 by mwittig