mwittig/node-milight-promise #30: Add support for 8-zone remotes

mwittig, 2017-10-09:
8-zone remotes appear to have a distinct command set. They are supported with newer makes of the ibox2 controller. See also https://github.com/dotsam/homebridge-milight/issues/39

fghekman, 2017-10-11:
Would be great if you manage to add support for 8-zones. When you need any additional information or you want me to test it just let me know.

mwittig, 2017-10-11:
@fghekman Just to give a brief update: I have analyzed a few packet capture files I have created including the datagrams sent by the Milight App with the 8-zone controller. While the general command layout appears to be the same as for the four channel controller (no suprise here) the command bytes for the target differ. I think I can quickly transfer the result into a first test application - maybe just to switch light on and off as a starting point. If this is going to be successful covering the whole command set will be pretty much straightforward. I'll get back to you soon.

mwittig, 2017-10-11:
@fghekman Do you have access to some Linux system for testing or do you need instructions for Windows?

fghekman, 2017-10-11:
Sounds indeed as a good starting point to just test the on/off state of the lights. 

Linux please. I have my homebridge config running on a Raspberry PI right now. I do have a spare PI so I could set it up as a test environment for this purpose.

mwittig, 2017-10-12:
@fghekman ok, great. here is a short recipe that should get you going. 

```
git clone https://github.com/mwittig/node-milight-promise.git
cd node-milight-promise
npm install 
cd examples
node example-8zone-v6
```

If you managed to execute the example program the lights should go on and off 4 times. If this is not the case you may need to edit `example-8zone-v6.js` with a text editor, e.g. vim or nano, to set the IP-address assigned to ibox2 wifi controller.

fghekman, 2017-10-15:
@mwittig Executed your code today and the lights did respond as you described. The 8 lights which are assigned to my iBox2 and the "virtual" 8-zone remote in the app did go on and off 4 times. I did test it with an other iBox2 and also these 8 lights did respond to the code. Seems like you did manage to track down the matching code for the target. (Did not need to set the IP-address in the .js file)

If you like any more info please let me know.

mwittig, 2017-10-16:
@fghekman That's great news. Thanks for testing! Next step for me is to add more command codes to the library and to extend the example program code accordingly. I'll get back to you in a few days for another tests. 

Once that's done I'll publish a new release and then, we can kindly ask @dotsam to integrate the new release into homebridge-milight which should only require minimal effort.

> Did not need to set the IP-address in the .js file

The example application used the global IP broadcast address. In many cases this works right away as the broadcast is directed to all hosts on the local network. However, on multi-homed hosts it depends on the routing configuration whether or not the packets will directed to ibox2 wifi controller. In such cases a network-specific broadcast address or the address of the ibox2 wifi controller is required.

fghekman, 2017-11-01:
@mwittig Just curious what the status is at the moment. Did you had the chance to dive into the library already? Please let me know if I can be of any help.

mwittig, 2017-11-02:
@fghekman Sorry for the delay. I had a few days off. Will look into this asap.

fghekman, 2017-11-23:
@mwittig Any news? Or are you still very busy at the moment.

mwittig, 2017-11-24:
Still busy, unfortunately. I have made some progress though and I'll push updates asap

pauleec, 2017-12-18:
@mwittig I have an 8-Zone setup here available I can test on if you need another tester. It would be good to see this enhancement in production as all the new RGB-CCT lights made after August now support operation in 8-zone configurations. Let me know how I can get involved and help.

mwittig, 2017-12-20:
@pauleec Thank you very much for the kind offer to help with testing!

@pauleec & @fghekman 
The required command sequences for 8-zone CCT color, white mode, brightness and effect mode control have landed on master which can be installed as follows:

```
git clone https://github.com/mwittig/node-milight-promise.git
cd node-milight-promise
npm install 
cd examples
node example-8zone-v6
```
If you managed to execute the example program the lights should go on and off followed by cycling the color. Then it should switch to white mode and cycle the white temperature. Finally the light should be slowly dimmed and then turned off . If this is not the case you may need to edit example-8zone-v6.js with a text editor, e.g. vim or nano, to set the IP-address assigned to ibox2 wifi controller.

pauleec, 2017-12-23:
@mwittig I tested your example as detailed and can confirm all 8 zones lights performed as expected. Thanks!

mwittig, 2017-12-23:
@pauleec Great. Thank you very much for testing. We may need to follow up on this to check the color mapping to make sure RGB or HSV are properly to the milight color circle. I'll publish a new release anyway asap

fghekman, 2017-12-24:
@mwittig Thank you for this new test update. I've tested the software also and can confirm it is operating as expected. Could not test it earlier as I had some days of and not near my house.
Looking forward to your next test release.

@pauleec Thank you for joining and testing this software also.

mwittig, 2017-12-27:
Released node-milight-promise@0.3.0. Thanks again for your testing support.

mwittig, 2017-12-27:
I am closing this now. Please re-open if you find problem with some of the control commands, e.g. color rendition

closed 2017-12-27 by mwittig