mwittig/node-milight-promise #32: Second bridge doesn't work

derzeitlose, 2017-10-19:
First Limitless LED bridge ibox2 works like a charm. Yesterday I added a second ibox2 bridge - nothing.

my config.jason:
{
                "platform":"MiLight",
                "name":"MiLight",
                "bridges": [
                        {
                        "ip_address": "192.168.10.105",
                        "version": "v6",
                        "lights": {"fullColor": ["Room1"],
                                "rgbw": ["Room1A","Room2","Room3"]},
                        "repeat": 1,
                        "delay": 100
                        }, {
                        "ip_address": "192.168.10.106",
                        "lights": {"rgbw": ["Test"]}
                        }

homebridge DEBUG:
[2017-10-19 22:43:03] [MiLight] [Test] Setting power state to on
[2017-10-19 22:43:03] [MiLight] [Test] Setting brightness to 100
[2017-10-19 22:43:03] [MiLight] [Test] Ommiting 'on' command as we've sent it to this bulb most recently
2017-10-19T20:43:03.426Z Milight: bytesSent=3, buffer=[0x45,0x00,0x55]
2017-10-19T20:43:03.536Z Milight: bytesSent=3, buffer=[0x4E,0x19,0x55]
[2017-10-19 22:43:05] [MiLight] [Test] Setting power state to off
2017-10-19T20:43:05.136Z Milight: bytesSent=3, buffer=[0x46,0x00,0x55]
[2017-10-19 22:38:04] [MiLight] [Room2] Setting power state to on
[2017-10-19 22:38:04] [MiLight] [Room2] Ommiting 'on' command as we've sent it to this bulb most recently
[2017-10-19 22:38:04] [MiLight] [Room2] Setting brightness to 100
2017-10-19T20:38:04.094Z Milight: bytesSent=22, buffer=[0x80,0x00,0x00,0x00,0x11,0x62,0x00,0x00,0x05,0x00,0x31,0x00,0x00,0x07,0x03,0x01,0x00,0x00,0x00,0x01,0x00,0x3D]
2017-10-19T20:38:04.149Z Milight: bytesReceived=8, buffer=[0x88,0x00,0x00,0x00,0x03,0x00,0x05,0x00], remote=192.168.10.105
2017-10-19T20:38:04.250Z Milight: ready for next command

If I add version v6 in the second bridge:
[2017-10-19 21:49:27] [MiLight] [Test] Setting power state to on
[2017-10-19 21:49:27] [MiLight] [Test] Setting brightness to 100
Unhandled rejection Error: no response timeout
    at Timeout._onTimeout (/usr/lib/node_modules/homebridge-milight/node_modules/node-milight-promise/src/milight-v6-mixin.js:123:26)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)

derzeitlose, 2017-10-19:
Found a solution:
Had to add Port 5987 for the second bridge in config.jason
But both milight bridges set to port 8899 in the bridge web interface

mwittig, 2017-10-19:
I am confused. You said you have two ibox2 bridges. However, you did not set the protocol ve4rsion to "v6" for the bridges as part of your json configuration file. If you do so it should not be required to set the port.

```
{
                "platform":"MiLight",
                "name":"MiLight",
                "bridges": [
                        {
                        "ip_address": "192.168.10.105",
                        "version": "v6",
                        "lights": {"fullColor": ["Room1"],
                                "rgbw": ["Room1A","Room2","Room3"]},
                        "repeat": 1,
                        "delay": 100
                        }, {
                        "ip_address": "192.168.10.106",
                        "version": "v6",
                        "lights": {"rgbw": ["Test"]}
                        }

derzeitlose, 2017-10-19:
And now again, after reboot the rasbery
Unhandled rejection Error: no response timeout

derzeitlose, 2017-10-19:
If I set version v6 - both Ports automatically set to 5987 - fine.
But after reboot, the response time out error is back again.

2017-10-19T23:07:46.465Z Milight: no response timeout
2017-10-19T23:07:46.543Z Milight: bytesSent=27, buffer=[0x20,0x00,0x00,0x00,0x16,0x02,0x62,0x3A,0xD5,0xED,0xA3,0x01,0xAE,0x08,0x2D,0x46,0x61,0x41,0xA7,0xF6,0xDC,0xAF,0xD3,0xE6,0x00,0x00,0xC9]
2017-10-19T23:07:47.555Z Milight: no response timeout
2017-10-19T23:07:47.568Z Milight: Session failed: Error: no response timeout
    at Timeout._onTimeout (/usr/lib/node_modules/homebridge-milight/node_modules/node-milight-promise/src/milight-v6-mixin.js:125:26)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)
Unhandled rejection Error: no response timeout
    at Timeout._onTimeout (/usr/lib/node_modules/homebridge-milight/node_modules/node-milight-promise/src/milight-v6-mixin.js:125:26)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)

Ok, here are my config.json

{
                "platform":"MiLight",
                "name":"MiLight",
                "bridges": [
                        {
                        "ip_address": "192.168.10.105",
                        "version": "v6",
                        "lights": {"fullColor": ["Room1"],
                                "rgbw": ["Room1A","Room2","Room3"]},
                        "repeat": 1,
                        "delay": 100
                        },
                        {
                        "ip_address": "192.168.10.106",
                        "version": "v6",
                        "lights": {"fullColor": ["Test6"],
                                "rgbw": ["Test1", "Test2", "Test3", "Test4"]},
                        "repeat": 1,
                        "delay": 100
                        }

derzeitlose, 2017-10-19:
OK. Full stop. After testing, try and error I found the reason for the error. Both ibox2 bridges are identical - did this in web configuration. Static IP, Changed from TCP to UDP. But the new ibox2 bridge, which I installed yesterday, is going in standby after a shot period of time and also shut down network connection. BAM! ... no response ...

mwittig, 2017-10-20:
Oh, I haven't see that before. Why should it go in stand-by?! These device are designed to be always on.

> Wake up ist only possible with iPhone App "Mi-Light".

What do you do to wake up the iBox with the App?

derzeitlose, 2017-10-27:
After ca. 20min the bridge is going offline until I start the app, then its possible to ping the bridge und open the web gui. I'll send the bridge back and try to get a new one. I think the bridge is faulty and have a bug.

closed 2017-10-27 by derzeitlose