mwittig/node-milight-promise #34: Raspberry pi

chriscartin, 2017-11-01:
Really wanted to run this on the pi.

Getting the error below, guessing its because the pi node version is so old (and cant be updated).

Any work around / fix?

Unhandled rejection TypeError: Object function Array() { [native code] } has no method 'from'
    at /home/promise/node_modules/node-milight-promise/src/milight-v6-mixin.js:134:42

mwittig, 2017-11-02:
@chriscartin Which version of nodejs do you run? You can just type `node --version` to find out. Note, oonly version 4.x and higher is supported by node-milight-promise

If you don't use nodejs for other applications my advice is to update your nodeJS setup. My advice is NOT to use the Raspbian apt package for nodejs as this will be most likely outdated. Remove the package and install the suitable binary package from https://nodejs.org/en/download/

chriscartin, 2017-11-02:
The issue is that the pi, by default comes with version v0.10.26. I eventually found a way to install a 'proper' version of node.js but that was only version 6. It seems that's the highest version supported by the ARM6 processor in my Pi so I gave up.
For anyone interested, the solution I eventually used to control my milight bulbs was running this python script. 
https://github.com/Strontvlieg/miLight-V6-iBox2-Domoticz

mwittig, 2017-11-02:
Well, I think the current node.js releases are still supported for ARM6. Please check https://nodejs.org/download/release/v8.9.0/ for available builds of the current LTS release. Later on I will also add further details to get you going

mwittig, 2017-11-04:
No follow up. Closing this now

closed 2017-11-04 by mwittig