mwittig/node-milight-promise #36: Missing brightness function for CCT (WhiteCommand)

markg85, 2017-12-14:
Hi,

Just curious where the brightness function is for the WhiteCommand object, it's not there.
I did notice that CCT bulbs in the milight app only do have a brightness up/down (not a slider) and that up/down is in WhiteCommand. Does this mean there simply is no brightness command with a given percentage for those bulbs?

mwittig, 2017-12-14:
For CCT brightness cannot be set directly to a relative value, e.g 50%, but you can increase or decrease the brightness (see also [examples folder](https://github.com/mwittig/node-milight-promise/tree/master/examples) and the code snippet below. Apart from this it is possible to switch to maximum brightness (`light.maxBright()`)

```
light.ready().then(function() {
  for (var x = 0; x < 10; x++) {
    light.sendCommands(commands.white.brightDown(zone));
    light.pause(100);
  }
  for (var x = 0; x < 10; x++) {
    light.sendCommands(commands.white.brightUp(zone));
    light.pause(100);
  }
...
```

mwittig, 2017-12-16:
For your application you can emulate setting the dim level, by calling `maxBright` followed by a number of `brightDown` calls to match the desired level. 

According to my findings there are 19 different dimming steps for CCTs supported by the milight controller. The number of `brightDown` calls required can be calculated as follows: 

`Round((100 - level) / 20) where level is a value in the range 0..100`. 

If the result is 20 you need to switch off the light. If you don't want to switch off lights at the lowest level use `Round((100 - level) / 20)` instead.

mwittig, 2017-12-18:
I am closing this now. Please re-open if you wish to follow up.

closed 2017-12-18 by mwittig