mwittig/node-milight-promise #40: Seeking help

haramhong, 2018-01-19:
Hi, I was wondering if I could seek some help with the node-milight-promise.

I have followed the guide (https://community.smartthings.com/t/milights-limitless-led-v6-habridge-node-google-home/95490/6) and managed to install NodeJS & HA-Bridge but encountered problem with node-milight-promise. Followings are the steps I took;

0. Setting up rasberry pie 3B, installed NodeJS & HA-Bridge.
    "node -v" returns v9.4.0.
    can access HA-Bridge Web Interface.

1. Installed node-milight-promise within "habridge" folder using "npm install node-milight-promise", which created "/habridge/node_modules/node-milight-promise".

2. Copied Yuriy's script (http://codecorner.galanter.net/2017/02/24/full-control-of-your-limitless-ledmilight-v6-bulbs-from-amazon-echo/) under "/habridge/led.js" for testing purposes.
    only changing ip to my Wifi Bridge ip address.

3. Testing "nodejs led.js 1 off" within "habridge" folder; Zone 1 off.

4. Received msg;
    "Invocation of asynchronous Milight commands done"
    "All command have been executed - closing Milight"

5. No response from LEDs.

*** I can control LEDs via app. I can scan WiFi Bridge via LimitlessLED admin tool for windows.
*** LimitlessLED Wifi Bridge, model no. iBox2 (the one w/ black strip, I am guessing this is V6?)
*** This is best I can provide in my knowledge and skills, please let me know if there is a log I could access & provide?

The Question is what am I doing wrong, and is there any way I can test if Wifi Bridge itself or node-milight-promise is working properly? I am not sure where it went wrong.

Is node-milihgt-promise sending commands properly?
Did I make mistake invoking commands?
Did I make mistake installing node-milight-promise?

Any help would be greatly appreciated~!

mwittig, 2018-01-30:
Sorry for the delay in reply! Did you manage to solve the issue, meanwhile= If not, I am happy to look into this, but it'll take some time before I can get back to you.

haramhong, 2018-01-31:
I have made a couple of attempt without luck. I think my knowledge on raspberrypie and nodes are too limited to set it up. (Any setup guide or tutorial would be much helpful).

I have found Milights-bridge which based on your nodes, and it has been successfully implemented. Although web interface was immensely helpful to test out the initial connection, I found HA-bridge commands are not as straight forward as milight-promise examples.

Again, if you have any step-by-step guide to direct me to, it would be greatly appreciated. Regardless, thank you for your wonderful works and reply!