mwittig/node-milight-promise #44: Add TCP-Support for v6.0

mwittig, 2018-03-06:
See https://github.com/dotsam/homebridge-milight/issues/51

mwittig, 2019-01-29:
> I have noticed that the UDP traffic generated by the script somehow blocks all UDP communication after a few days. The only solution after that seems to be reboot of the router. 

I have never seen this kind of behaviour. What kind of router do you have? I guess it is bug of the router firmware.

> Is there some documentation how the TCP commands would work? I havn't been able to find any documentation.

No, this is not documented as far as I know. At least, there is no official doc publically available.

> Is there some documentation how the TCP commands would work? I havn't been able to find any documentation on this. It seems to be straightforward with the legacy bridge, but I haven't seen examples for the v6 bridge

It may be a work-around in your case. On the other hand TCP might bring a bunch of new issues
* how to hhandle tcp connection in line with the bridge implementation
* if tcp connection are short-lived this may cause significant delay
* UDP-based applications will no longer work.

> When do you plan to add the TCP support? 

I don't have a timeline for this there has not been a lot of demand for this so far

fzmemon, 2019-01-29:
I have this router: https://www.tp-link.com/en/products/details/cat-9_Archer-C50.html
May be it is a router issue but I am not so sure. This is the error that I see from the script when a reboot of the router becomes necessary:
31-12-2018 21:41:41: Error: no response timeout