mwittig/node-milight-promise #47: node-milight-promise not controlling bridge

pauleec, 2018-04-30:
@mwittig Hi Marcus, I have spent a few nights trying to isolate a problem I am experiencing setting up control between the node-milight-promise module and a v6 iBox2 Milight Lamp/Bridge. The discovery script returns the correct information re. the v6 version and its ip but other than that nothing seems to be working - Ive tried each example command.

Can you confirm if there needs to be anything specific in the iBox2 configuration itself. The firmware of the iBox2 bridge is 1.008 web 1.012. Its configured to operate in STA mode and I have tried both UDP and TCP-Server and TCP-Client (?) configurations (neither work). The port listed in the web config for the bridge is 8899.

when i run any of the example scripts I get a response 

"Invocation of asynchronous Milight commands done
All command have been executed - closing Milight"

So its nice getting those messages - but it would be even nicer if something else happened but nothing is :)

I am running the latest version of node-milight-promise.

I'm stuck! What next?

Thx Pauleec

pauleec, 2018-04-30:
@mwittig Just to add to the last comment, I am using @dotsam 's homebridge-milight plugin. Homebridge is launching and with the debug option switched on I get the following in the console:

orangepi@OrangePI:~$ MILIGHT_DEBUG=true homebridge -D
*** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
*** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
[2018-4-30 14:37:26] Loaded plugin: homebridge-alexa
[2018-4-30 14:37:26] Registering platform 'homebridge-alexa.Alexa'
[2018-4-30 14:37:26] ---
[2018-4-30 14:37:29] Loaded plugin: homebridge-broadlink-rm
[2018-4-30 14:37:29] Registering platform 'homebridge-broadlink-rm.BroadlinkRM'
[2018-4-30 14:37:29] ---
[2018-4-30 14:37:29] Loaded plugin: homebridge-milight
[2018-4-30 14:37:29] Registering platform 'homebridge-milight.MiLight'
[2018-4-30 14:37:29] ---
[2018-4-30 14:37:29] Loaded plugin: homebridge-server
[2018-4-30 14:37:29] Registering platform 'homebridge-server.Server'
[2018-4-30 14:37:29] ---
[2018-4-30 14:37:29] Loaded config.json with 0 accessories and 4 platforms.
[2018-4-30 14:37:29] ---
[2018-4-30 14:37:29] Loading 4 platforms...
[2018-4-30 14:37:29] [Alexa] Initializing Alexa platform...
[2018-4-30 14:37:29] [MiLight] Initializing MiLight platform...
2018-04-30T14:37:29.861Z Milight: {"ip":"10.1.1.188","port":5987,"delayBetweenCommands":100,"commandRepeat":1,"fullSync":false,"sendKeepAlives":true,"sessionTimeout":30000,"host":"10.1.1.188","type":"v6","broadcastMode":false}
[2018-4-30 14:37:29] [MiLight] Initializing platform accessory 'Kitchen Downlighters'...
[2018-4-30 14:37:29] [MiLight] Initializing platform accessory 'Bridge Light'...
[2018-4-30 14:37:29] [Broadlink RM] Initializing BroadlinkRM platform...
[2018-4-30 14:37:29] [Broadlink RM] [INFO] Automatically discovering Broadlink RM devices.
[2018-4-30 14:37:30] [Broadlink RM] No accessories have been added to the "Broadlink RM" platform config.
[2018-4-30 14:37:30] [Broadlink RM] Initializing platform accessory 'Learn'...
[2018-4-30 14:37:30] [Broadlink RM] Initializing platform accessory 'Scan Frequency'...
[2018-4-30 14:37:30] [Homebridge Server] Initializing Server platform...
[2018-4-30 14:37:30] Loading 0 accessories...
[2018-4-30 14:37:31] [Homebridge Server] is listening on: http://10.1.1.76:8765
[2018-4-30 14:37:31] Homebridge is running on port 51826.
2018-04-30T14:37:31.624Z Milight: initializing socket
2018-04-30T14:37:31.668Z Milight: initializing socket done
[2018-4-30 14:37:31] [Broadlink RM] 
[2018-4-30 14:37:31] [Broadlink RM] **************************************************************************************************************
[2018-4-30 14:37:31] [Broadlink RM] ** Welcome to version 3.5.5 of the Homebridge Broadlink RM Plugin!
[2018-4-30 14:37:31] [Broadlink RM] ** 
[2018-4-30 14:37:31] [Broadlink RM] ** Find out what's in the latest release here: https://github.com/lprhodes/homebridge-broadlink-rm/releases
[2018-4-30 14:37:31] [Broadlink RM] ** 
[2018-4-30 14:37:31] [Broadlink RM] ** If you like this plugin then please star it on GitHub or better yet
[2018-4-30 14:37:31] [Broadlink RM] ** buy me a drink using Paypal https://paypal.me/lprhodes or crypto https://goo.gl/bEn1RW.
[2018-4-30 14:37:31] [Broadlink RM] ** 
[2018-4-30 14:37:31] [Broadlink RM] ** Keep up to date with this plugin along with everything HomeKit and homebridge
[2018-4-30 14:37:31] [Broadlink RM] ** by signing up to my newsletter at http://workswith.io
[2018-4-30 14:37:31] [Broadlink RM] **
[2018-4-30 14:37:31] [Broadlink RM] ** You can disable this message by adding "hideWelcomeMessage": true to the config (see config-sample.json).
[2018-4-30 14:37:31] [Broadlink RM] **
[2018-4-30 14:37:31] [Broadlink RM] **************************************************************************************************************
[2018-4-30 14:37:31] [Broadlink RM] 
2018-04-30T14:37:31.750Z Milight: Establishing new session
2018-04-30T14:37:31.774Z Milight: bytesSent=27, buffer=[0x20,0x00,0x00,0x00,0x16,0x02,0x62,0x3A,0xD5,0xED,0xA3,0x01,0xAE,0x08,0x2D,0x46,0x61,0x41,0xA7,0xF6,0xDC,0xAF,0xD3,0xE6,0x00,0x00,0xC9]
2018-04-30T14:37:32.002Z Milight: bytesReceived=22, buffer=[0x28,0x00,0x00,0x00,0x11,0x00,0x02,0xF0,0xFE,0x6B,0x1E,0x29,0x98,0x72,0x1D,0xE4,0xE2,0x00,0x01,0x0C,0x00,0x00], remote=10.1.1.188
2018-04-30T14:37:32.133Z Milight: ready for next command
2018-04-30T14:37:32.138Z Milight: Session Id: 0x0C,0x00
[2018-4-30 14:37:32] [Broadlink RM] [INFO] Discovered Broadlink RM2 Pro Plus (272a) at 10.1.1.22 (0d:43:b4:fc:1e:9c)
2018-04-30T14:37:32.259Z Milight: Session initialized
2018-04-30T14:38:02.265Z Milight: bytesSent=27, buffer=[0x20,0x00,0x00,0x00,0x16,0x02,0x62,0x3A,0xD5,0xED,0xA3,0x01,0xAE,0x08,0x2D,0x46,0x61,0x41,0xA7,0xF6,0xDC,0xAF,0xD3,0xE6,0x00,0x00,0xC9]
2018-04-30T14:38:02.381Z Milight: bytesReceived=22, buffer=[0x28,0x00,0x00,0x00,0x11,0x00,0x02,0xF0,0xFE,0x6B,0x1E,0x29,0x98,0x72,0x1D,0xE4,0xE2,0x00,0x01,0x2C,0x00,0x00], remote=10.1.1.188
[... cut ...]
2018-04-30T14:41:02.275Z Milight: bytesSent=27, buffer=[0x20,0x00,0x00,0x00,0x16,0x02,0x62,0x3A,0xD5,0xED,0xA3,0x01,0xAE,0x08,0x2D,0x46,0x61,0x41,0xA7,0xF6,0xDC,0xAF,0xD3,0xE6,0x00,0x00,0xC9]
2018-04-30T14:41:02.454Z Milight: bytesReceived=22, buffer=[0x28,0x00,0x00,0x00,0x11,0x00,0x02,0xF0,0xFE,0x6B,0x1E,0x29,0x98,0x72,0x1D,0xE4,0xE2,0x00,0x01,0xF8,0x00,0x00], remote=10.1.1.188
2018-04-30T14:41:02.555Z Milight: ready for next command
2018-04-30T14:41:02.556Z Milight: Session Id: 0xF8,0x00

So there seems to be communication between the homebridge-milight plugin and the iBox2 bridge - but commands sent are received by the iBox but are ignored...? Whats going on?

pauleec, 2018-05-03:
Fixed it! 

after lots of head scratching the issue finally came down to my friend who had set a password on the milight bridge. resetting back to the default (0000) and the devices all started working! If there is a tip in the FAQ I wish I had seen this earlier :) otherwise there would certainly be lots of time saved having one on this!

closed 2018-05-03 by pauleec

mwittig, 2018-05-04:
Sorry for delay in reply to your ticket. I have been very busy the last couple of days. I am glad you have managed solve the problem yourself! Actually, I have not been aware about this password issue and I certainly will make a troubleshooting note about that. 

How did you set the password the password and what it is exactly about? With my iBox1 there is no password for the bridge othert than the user account for the web interface and the WiFi credentials. Maybe you can post a screenshot of the UI where the password is set?