mwittig/node-milight-promise #52: Support Question

gemmi-arts, 2018-10-18:
Hello,

Does this module support the following controller?

[image]

mwittig, 2018-10-20:
Most likely it does, but I can't tell you for sure as I don't have this device at hand for testing. Note, you'll also need a Milight  Wifi Box to use the controller along with node-milight-promise (or generally to be able to control it with an Internet device like a Smartphone etc). Please let me know about your findings in case you'll give this a try. Thanks

mwittig, 2018-10-23:
Btw, do you really want to use an RGB controller? 

My recommendation is to use a RGBW controller. This is known to work with `node-milight-promise`  and you can use either RGBW or RGB stripes. The RGB+CCT controller is another option to look at. The price difference for these controller is marginal.

mwittig, 2018-10-29:
I am closing this now. Please re-open if you wish to follow up on this

closed 2018-10-29 by mwittig