mwittig/node-milight-promise #55: Inconsistent colors across commands

theaceofthespade, 2018-11-24:
First of all, the library is great, thanks so much!

I've been playing around with the different command sets (fullcolor vs fullcolor8zone specifically) and have noticed some inconsistencies in the coloring. On fullcolor, hue 0 is about 10 points closer to blue than it's supposed to be, however on fullColor8Zone, hue 42 (yellow) is about 10 points too green.  I've tried on several models of bulbs.

I'll be happy to jump in and help debug this, but figured I'd make sure it wasn't a known (or even expected) behavior first.

mwittig, 2018-11-28:
Thank your very much for your positive feedback and issue report. 

The problem is the vendor has implemented varying color circles with different control models.  I have tried to fix that for RGBWW versus full color, but I haven't done this for 8-zone as I am not able to test it.  I'll try to setup a test branch with 8-zone fixture based on your findings. I'll keep you posted.

theaceofthespade, 2018-11-29:
I appreciate it!  For the other ones, was it something you were able to test programatically, or did you have to eyeball the color differences?