mwittig/node-milight-promise #56: On/Off does not work with RGBW One Zone Dimmer

pa-re, 2018-12-09:
I've got a version of the milight RGBW dimmers that let's me control RGB and W on the LED strips at the same time. This dimmer and remote combination does only support one zone as also depicted in the milight app.

I got it to work in the ioBroker.milight adapter with most of the functionality, except On/Off which would be quite handy for obvious reasons. I tried to figure out the necessary command sequence changes myself, but I cannot find the specific ones for these devices.

**My settings:**
milight box version/port: v6, port 5987
Lamp Zone 1: RGBW (RGBWW does not work at all)
I did not try other zones.

Delay 100, Repeat 1

**Screenshots of the hardware:**
[image]

[image]

**The question is**: Can you help me figure out the correct command sequence for this particular RGBW dimmer?

Best Regards,
Patrick

mwittig, 2019-01-03:
Sorry for the delay in reply. I can have a look into this now if you are still working on this. Did you make progress on this since you opened this issue? 

The RGBW remote you are trying to use is currently not supported, but I am happy to look into this. So far, I only have support for the single channel RGB controller.

pa-re, 2019-01-03:
Dear Mr Wittig,

thank you for the response and the offer to look into it. I am still very interested in it as my work on a solution and workarounds failed so far. I got some more information, but always hit a roadblock.

I can confirm that your implementation is the only one that successfully creates a "link" and thus works (mostly) with the FUT028. The content of "commandsV6.js" after
// RGBWW commands
RgbwCommand [...]
works except the on/off.

I tried **Wireshark,** but the package scrambling stopped me to discover the real sequence of the FUT028 (Dimmer Model). I tried reversing the checksum and scrambling sequence, but unfortunately I couldn't read your Javascript nor the C and C++ implementations I found online to actually reverse it. That said I could still capture some commands with wireshark and decrypt it, but I'd need help to actually make the messages readable. Do you maybe already have a "wireshark decryption module"?

How can I help you to help me?

Best Regards,
Patrick

mwittig, 2019-01-23:
Patrick, I am not aware of a Wireshark dissector for the Milight protocol. However, as the UDB datagrams exchanged by milight are not encrypted it is possible to determine the command sequences. Possibly, filtering should be used, i.e. UDP to traffic sent/targeted at the iBox IP on port 5987.

Questions: 

- Are you able to control ALL functions including on/off using the Milght App with the control you have included in the first post?
- Did you try the "rgb" command set of the command API?

pa-re, 2019-01-26:
Hi Marcus,

I do have some wireshark records for the on/off from the app. Is that helpful to you?

To your questions:

- Yes, the app let's me select that remote and it works with all functions of the FUT028. That's the wireshark records I captured and could provide to you.
- You gave me a nice idea. I replaced the on/off commands in the JS file and tried all the ones I listed below. Unfortunately without success. I also tried some arbitrary changes to those codes in the hopes to find the right one. Unfortunately it didn't work.

=> I am out of ideas. How can I help you?

zoneCtlRGBWFactory
on: function () {
    return [0x31, 0x00, 0x00, 0x05, 0x02, 0x09, 0x00, 0x00, 0x00, zoneID];
},
off: function () {
    return [0x31, 0x00, 0x00, 0x05, 0x02, 0x0a, 0x00, 0x00, 0x00, zoneID];
},

zoneCtlRGBWWFactory
on: function () {
    return [0x31, 0x00, 0x00, 0x08, 0x04, 0x01, 0x00, 0x00, 0x00, zoneID];
},
off: function () {
    return [0x31, 0x00, 0x00, 0x08, 0x04, 0x02, 0x00, 0x00, 0x00, zoneID];
},

zoneCtlRGBWFactory
on: function () {
    return [0x31, 0x00, 0x00, 0x07, 0x03, 0x01, 0x00, 0x00, 0x00, zoneID];
},
off: function () {
    return [0x31, 0x00, 0x00, 0x07, 0x03, 0x02, 0x00, 0x00, 0x00, zoneID];
},

zoneCtlRGBFactory
on: function () {
    return [0x31, 0x00, 0x00, 0x05, 0x02, 0x09, 0x00, 0x00, 0x00, zoneID];
},
off: function () {
    return [0x31, 0x00, 0x00, 0x05, 0x02, 0x0a, 0x00, 0x00, 0x00, zoneID];
},

zoneCtlWhiteFactory
on: function () {
    return [0x31, 0x00, 0x00, 0x01, 0x01, 0x07, 0x00, 0x00, 0x00, zoneID];
},
off: function () {
    return [0x31, 0x00, 0x00, 0x01, 0x01, 0x08, 0x00, 0x00, 0x00, zoneID];
},

baseCtlFactory
on: function () {
    return [0x31, 0x00, 0x00, 0x00, 0x03, 0x03, 0x00, 0x00, 0x00, zoneID];
},
off: function () {
    return [0x31, 0x00, 0x00, 0x00, 0x03, 0x04, 0x00, 0x00, 0x00, zoneID];
},

BridgeLEDCommands.prototype.on = function() {
  return [0x31, 0x00, 0x00, 0x00, 0x03, 0x03, 0x00, 0x00, 0x00, 0x01]
};

BridgeLEDCommands.prototype.off = function() {
  return [0x31, 0x00, 0x00, 0x00, 0x03, 0x04, 0x00, 0x00, 0x00, 0x01]
};

RgbwCommand.prototype.on = function(zone) {
  var zn = Math.min(Math.max(zone, 0x00), 0x04);
  return [0x31, 0x00, 0x00, 0x07, 0x03, 0x01, 0x00, 0x00, 0x00, zn]
};

RgbwCommand.prototype.off = function(zone) {
  var zn = Math.min(Math.max(zone, 0x00), 0x04);
  return [0x31, 0x00, 0x00, 0x07, 0x03, 0x02, 0x00, 0x00, 0x00, zn]
};

WhiteCommand.prototype.on = function(zone) {
  var zn = Math.min(Math.max(zone, 0x00), 0x04);
  return [0x31, 0x00, 0x00, 0x01, 0x01, 0x07, 0x00, 0x00, 0x00, zn]
};

WhiteCommand.prototype.off = function(zone) {
  var zn = Math.min(Math.max(zone, 0x00), 0x04);
  return [0x31, 0x00, 0x00, 0x01, 0x01, 0x08, 0x00, 0x00, 0x00, zn]
};

RgbFullColorCommand.prototype.on = function(zone) {
  var zn = Math.min(Math.max(zone, 0x00), 0x04);
  return [0x31, 0x00, 0x00, 0x08, 0x04,0x01, 0x00, 0x00, 0x00, zn]
};

RgbFullColorCommand.prototype.off = function(zone) {
  var zn = Math.min(Math.max(zone, 0x00), 0x04);
  return [0x31, 0x00, 0x00, 0x08, 0x04, 0x02, 0x00, 0x00, 0x00, zn]
};

RgbCommand.prototype.on = function() {
  return [0x31, 0x00, 0x00, 0x05, 0x02, 0x09, 0x00, 0x00, 0x00, 0x01]
};

RgbCommand.prototype.off = function() {
  return [0x31, 0x00, 0x00, 0x05, 0x02, 0x0a, 0x00, 0x00, 0x00, 0x01]
};

pauleec, 2019-01-27:
Hi guys, I also have a number of these led strip drivers which were part of a kit which sold the remote and strip driver together. I tried to link the strip driver to an 8 and 4 zone remote which had been driving rgb-cct 8 zone leds.... and was unable to link or connect . This was the same when I attempted to pair with the Mi-Light bridge. I spoke to Futlight who informed me that the kit units were using. Different incompatible protocol. i have 10 sets of the kits and have subsequently only used them when all the install requires is a single remote control strip, it would be awesome to find a way to synch the strip driver to the bridge but so far I’ve not been able to do it. I’m building an RFLink to see if i can understand a little more about what’s goin on. If u wud like any additional testing I can help...

mwittig, 2019-01-27:
@pa-re Thank you very much for investigating this further. Meanwhile I amnged to sniff bits of the command set used by the single zone remote (fut027).. For this purpose I have been using the packet capture feature which is bulit into my access router. The current master contains an experimental extension to the command set for single zone. 

``` 
var Milight = require('../src/index');
var commands = Milight.commandsV6;

var light = new Milight.MilightController({
    ip: "255.255.255.255",
    type: 'v6'
  }),
  zone = 1;

light.ready().then(function() {

  light.sendCommands(commands.rgbwSingle.on(),  commands.rgbwSingle.brightness(100));
  light.pause(1000);

  light.sendCommands(commands.rgbwSingle.off());
  light.pause(1000);

}).catch(function(error) {
  console.log(error);
  light.close();
});

mwittig, 2019-01-27:
@pauleec I am not sure whether or not I understood you previous post correctly. Are you able to link and control your set with Milight Smartphone App using the Single Zone remote? 

- If yes, I can confirm it is a different command set and most likely it is not possible to link the bulbs/strip controllers with an 8-zone or 4-zone remote at all. It might be possible with Milight Bridge project based on ESP8266, but I am not sure about this.
- If no, are you able to control it all with the Milight Smartphone App?

https://github.com/sidoh/esp8266_milight_hub

pa-re, 2019-01-28:
Hi Marcus,

seems you did it. I only had a couple of minutes to try out the example script you provided, but it switches the lights on and off successfully. I did not try any other functions, yet.

Best Regards,
Patrick

pa-re, 2019-01-28:
Hi Marcus,

turns out your script turns on all and any RGBWSingle light. Is there any way to only bind/link certain lights to the bridge?

Before the test I had only one strip linked to the bridge and to iobroker. IOBroker doesn't work anymore, but your script activates all lights. I do not know if this is still related to my original request or if I am misusing your tool in general.

Best Regards,
Patrick

mwittig, 2019-01-29:
@pa-re That's great news!

> turns out your script turns on all and any RGBWSingle light. Is there any way to only bind/link certain lights to the bridge?

How does this behave when you control the lights with Milight Smartphone App? I assume it also switches all lights right now. If so, try to unlink a light from the bridge as follows:
* turn of the light/strip controller for at least 3 seconds
* turn it on again and press unlink button of the Milight App 
* as a feedback the light/led strip should blink now 5 times. 

Btw, this is a bit of guess work as I have no device to test with. I have RGBW bulbs and according to the docs it should be possible to use them with single zone controller, but it does not work with my iBox 1 controller.

pa-re, 2019-02-10:
Hi Marcus,

I followed your advice and you were right. I must have linked both strips to the bridge at some point in the past. Now on/off is working as expected.

On a slightly different subject: As far as I understand the milight protocol only implements exactly what the remotes can do and nothing more, right? So there's no way to set the white brightness to an exact level, but only do brighter/darker as the remote can do. That's unfortunate, but I built a way around it with a persistent variable, but that means I cannot use the whitechannel on the remote anymore as I have no way to know the current state of the strip then.

Functions that are tested and work for me:
- On/off
- Whitechannel brighter/darker via whiteMode/nightMode
- Setting a color
- Changing the brightness of the color

Buttons from the remote that I couldn't get to work so far:
- r/g/b up/down buttons that end up at white when all are maxed out.

Thank you a lot for your help. I think I'll somehow get the sunrise wakeup realized with that functionality.

Do you want to continue to support everything/every button or call it quits?

Best Regards,
Patrick

mwittig, 2019-02-11:
Hi Patrick,

thanks for your feedback. Please find my comments below.

> As far as I understand the milight protocol only implements exactly what the remotes can do and nothing more, right?

Yes, that's right. 

> So there's no way to set the white brightness to an exact level, but only do brighter/darker as the remote can do

It depends on the type of remote control, really. The older controllers (single-zone controllers, controller for white bulbs) only support brightness up/down. The newer ones, like the 4-zone RGBW controller support brightness values. Apparently, it is only possible to link a bulb/leed controller to a specific remote control, most of the time. That's at least what I found out with the few Milight devices I have.

> That's unfortunate, but I built a way around it with a persistent variable, but that means I cannot use the whitechannel on the remote anymore as I have no way to know the current state of the strip then.

What I can do is to add is four functions to control white, red, green, and blue color individually. I am not sure, how this works, as I haven't decoded this part of the protocol, yet. It might be a way of setting the brightness of the four base colors to a given value.

mwittig, 2019-02-13:
I have decoded the missing bits for the single zone controller to chnage the color channels individually. Unfortunately, it only support up/down like it is the case for brightness control. I'll add this this to the command interface, though

mwittig, 2019-03-05:
Closing this now. Please re-open if you wish to follow-up

closed 2019-03-05 by mwittig