mwittig/node-milight-promise #58: color

Derrin616, 2018-12-21:
how can i set the color to blue / red etc i can't find it 
im using your NPM package for javascript

mwittig, 2019-01-03:
Hi, thanks for using node-milight-promise!

The command library provides the `rgb()` function to set the RGB color.  For RGBW and full color bulbs the function expects four numeric parameters. The first parameter sets the Milight zone, while the latter parameters provide the color value for three RGB channels red, green, and blue. Each RGB channel value can range from `0..255`. See example below.

Note, however, as indicated in the README the color space of Milight of the legacy RGBW bulbs is limited, as it is not possible to control color saturation and there are only 20 brightness levels. Thus, the results may be disappointing when compared to other LED lightning technologies. If you are looking for a more convenient way of applying color values have a look at the [color](https://www.npmjs.com/package/color) which provides an easy way of mapping CSS color strings to RGB values.

``` 
# set color blue
light.sendCommands(commands.fullColor.rgb(zone 0, 0, 255));

Derrin616, 2019-01-03:
So if i want red i need to do this

light.sendCommands(commands.fullColor.rgb(zone 255, 0, 0));

mwittig, 2019-01-04:
Yes, that's it!

As said in my previous post you may want to look into the [color](https://www.npmjs.com/package/color) npm package which provides an easy way of mapping CSS color strings to RGB values. See the following snippet, for example:

``` 
var Color = require('color');
var color = Color("lightgreen");

light.sendCommands(commands.fullColor.rgb(zone, color.red(), color.green(), color.blue()));

mwittig, 2019-01-09:
Setting this to closed for now. Please re-open if you wish to follow up on this

Derrin616, 2019-01-10:
How can i change the color of i box i tried many things

    light.sendCommands(commands.fullColor.bridge.rgb(zone, color.red(), color.green(), color.blue()));
    light.sendCommands(commands.bridge.fullColor.rgb(zone, color.red(), color.green(), color.blue()));

and more i can't find it 

Regards
Derrin616

mwittig, 2019-01-11:
See the examples provided in the [examples directory](https://github.com/mwittig/node-milight-promise/tree/master/examples).  Now the question is which bulb type you have.
* If you have a RGB+CCT bulb which provides for changing the color/white temperature (white color from warm white to cold white) have a look at example-full-colour.js
* If you a RGBWW bulb which does not provide for changing the color/white temperature see example-rgbw-v6.js

I hope this helps.  If not, please providedescription of you test setup. Thanks

Derrin616, 2019-01-11:
I cant find how change rhe color of the bridge in the fullcolor.js

mwittig, 2019-01-11:
If you start with example-full-colour.js example the command is as follow below. I am not sure where the "bridge" component comes from you have included in your earlier post.

```
light.sendCommands(commands.fullColor.rgb(zone, color.red(), color.green(), color.blue()));

Derrin616, 2019-01-11:
i can control the color of my led strip noiw but i want to control the color of my bridge / ibox i hope you understand me now XD

mwittig, 2019-01-11:
Ah, ok. Sorry, for the misunderstanding. To control the bridge light color, please have a look at the example `example-bridge-v6.js`. The color command is as follows:

```
light.sendCommands(commands.bridge.rgb(zone, color.red(), color.green(), color.blue()));

Derrin616, 2019-01-13:
Hello sorry for opening it again XD it works fine now but the color does not work as i want 

I used the color npm package and i made a program in vb.net to control the milight 

if i change the ibox color to red my program will write the hex color into a json file

this works fine for my ledstrips but if i change the color to red in my program the ibox color will be green 

super weird do you have any idea why this is like this because the ledstrip works fine

My codes:

json file:

{"prefix":"#FF0000"}

my javascript file:

var Milight = require('../src/index');
var commands = Milight.commandsV6;
const ip = require('./ip.json')
const z = require('./zone.json')
const col = require('./iboxcolor.json')
var Color = require('color');
var color = Color(col.prefix);

// Important Notes:
// *  Instead of providing the global broadcast address which is the default, you should provide the IP address
//    of the Milight Controller for unicast mode. Don't use the global broadcast address on Windows as this may give
//    unexpected results. On Windows, global broadcast packets will only be routed via the first network adapter. If
//    you want to use a broadcast address though, use a network-specific address, e.g. for `192.168.0.1/24` use
//    `192.168.0.255`.
// *  Note, for the V6 command set each command must be provided with a zone parameter as shown in the example below!

var light = new Milight.MilightController({
    ip: ip.prefix,
    type: 'v6'
  }),
  zone = z.prefix;

light.ready().then(function() {

    light.sendCommands(commands.bridge.rgb(zone, color.red(), color.green(), color.blue()));

  light.close().then(function () {
    console.log("sluiten");
  });
  console.log("klaar");
}).catch(function(error) {
  console.log(error);
  light.close();
});

I hope you can fix the issue i think this is a problem with your npm package

Please tell me if you have questions or if i need to explain it better

Regards 
Derrin

mwittig, 2019-01-15:
Omg, sometimes I am really stupid. Sorry about that. 

With the bridge light there is no zone parameter! If you remove it from the call it'll work as expected. 
I have corrected my previous post, accordingly.

```
light.sendCommands(commands.bridge.rgb(color.red(), color.green(), color.blue()));

Derrin616, 2019-01-15:
Haha no problem thanks for the help i hope it works

Derrin616, 2019-01-15:
It works fine

Conclusion:

//set color for a zone
light.sendCommands(commands.fullColor.rgb(zone, 255, 0, 0));//color red

//simpler version with color module
var Color = require('color');
var color = Color("lightgreen");// the color that you want you can also use hex color

light.sendCommands(commands.fullColor.rgb(zone, color.red(), color.green(), color.blue()));

//set color for the ibox - bridge
light.sendCommands(commands.bridge.rgb(255, 0, 0));//color red

//simpler version with color module
var Color = require('color');
var color = Color("lightgreen");// the color that you want you can also use hex color
light.sendCommands(commands.bridge.rgb(color.red(), color.green(), color.blue()));

closed 2019-01-15 by Derrin616