mwittig/node-milight-promise #59: Help to user legacy bridge with command2

kekec777, 2019-01-03:
Hallo everyone 

Im quite new to this and i need some help

I have a old milight bridge and it seems i cannnot get this to work with just installing and running node server.js

so i tried to make a test.js with this inside:
var Milight = require('node-milight-promise').MilightController;
var commands = require('node-milight-promise').commands2;

// Important Notes:
// Instead of providing the global broadcast address which is the default, you should provide the IP address
// of the Milight Controller for unicast mode. Don't use the global broadcast address on Windows as this may give
// unexpected results. On Windows, global broadcast packets will only be routed via the first network adapter. If
// you want to use a broadcast address though, use a network-specific address, e.g. for `192.168.0.1/24` use
// `192.168.0.255`.

var light = new Milight({
		ip: "255.255.255.255",
		delayBetweenCommands: 75,
		commandRepeat: 2
	}),
	zone = 1;

light.sendCommands(commands.rgbw.on(zone), commands.rgbw.whiteMode(zone), commands.rgbw.brightness(100));
light.pause(1000);

light.sendCommands(commands.rgbw.off(zone));
light.pause(1000);

// Setting Hue
light.sendCommands(commands.rgbw.on(zone));
for (var x = 0; x < 256; x += 5) {
	light.sendCommands(commands.rgbw.hue(x));
	if (x === 0) {
		commands.rgbw.brightness(100)
	}
	light.pause(100);
}
light.pause(1000);

light.sendCommands(commands.rgbw.off(zone));
light.pause(1000);

// Back to white mode
light.sendCommands(commands.rgbw.on(zone), commands.rgbw.whiteMode(zone));
light.pause(1000);

// Setting Brightness
light.sendCommands(commands.rgbw.on(zone));
for (var x = 100; x >= 0; x -= 5) {
	light.sendCommands(commands.rgbw.brightness(x));
	light.pause(100);
}
light.pause(1000);

light.sendCommands(commands.rgbw.off(zone));
light.pause(1000);

light.close().then(function () {
	console.log("All command have been executed - closing Milight");
});
console.log("Invocation of asynchronous Milight commands done");

and this is working :) - hurray (the led's changing colors)
So how can i include this instead ???

Please help a newbie :)

thanks in advise

mwittig, 2019-01-03:
Hi, thanks for using node-milight-promise! 

I am not sure what do you mean by running ```node server.js```! Is ```server.js``` part of some source code you have written or do you use some other application such as homebridge-milight which incorporates  node-milight-promise? 

Please note, node-milight-promise provides an API developers can use to send Milight commands as part of their application. It does not provide a command line interface or generic server so far.

mwittig, 2019-01-04:
> Then into folder node_modules and then I run npm install node-miilight-promise

Did you also install the other dependencies? Generally, it should be ok to simply execute `npm install` with further parameters as npm will pick-up the dependencies from the package descriptor. The following works for me

```
git clone https://github.com/KevinVR/milights-bridge.git
cd milights-bridge/
npm install
cp config.example.js config.js
# setup config 
vi config.js
node server.js
```

`milights-bridge` works for me with the new v6 controller, however it does work with v4! Having investigated the source code I can tell you that the code of `milights-bridge` does **not** work with v4 although the author is suggesting it works as implemented.  Are you familiar with Javascript? If yes I can roughly tell you which changes are required to fix it.

agoralive, 2019-01-15:
I am in the same situation as kekec777, is there a way to type a command which allows me to control my bulbs on zone 5 in an 8 zone remote?

mwittig, 2020-01-14:
Sorry, guys I currently have no time to make a contribution to the milight-controller project myself. My suggestion is to raise an issue at https://github.com/KevinVR/milights-bridge/issues

closed 2020-01-14 by mwittig