mwittig/node-milight-promise #65: changing the delay before light turns on after hitting a switch

tb91, 2019-05-19:
Hey guys,
I did not know where to post this so I am hoping that you won't be mad and even might have a solution.
Right now when I turn on my milights with my phone it works instantaneously. But as soon as I use a switch to turn on the "electricity" it waits roughly 200ms before it turns on. Somewhere I read that this is intended and is used to "listen" for a command after a power cut or something. However, I don't want this delay. Anybody experienced the same or know a solution to this?
Regards.
Tim

mwittig, 2019-05-23:
Thank you very much for using node-milight-promise!

The delay before the bulb turns on is a sort of pairing mode which is used to link/unlink controllers/bridges with/from the bulb. As far as I know this value cannot be changed. It is hard-wired in the firmware of the bulbs.

mwittig, 2020-01-09:
Closing this as answered

closed 2020-01-09 by mwittig