mwittig/node-milight-promise #66: How do I change the Color temperature?

mx-web, 2019-05-23:


mwittig, 2019-05-23:
Thank you very much for using node-milight-promise! 

Am I right in assuming you have a Dual-White Bulb? In this case please have a look at [example-dual-white-v6.js](https://github.com/mwittig/node-milight-promise/blob/master/examples/example-dual-white-v6.js#L43-L51). The command functions are `warmer(zone)` and `cooler(zone)` which increase/decrease the white color temperature. Note, the milight firmware does not provide for setting an absolute temperature value at this point. I am also assuming here you have an iBox 1 or 2 bridge. If you own an older milight bridge have a look at example-dual-white.js for the the legacy interface.

In case you have a Full Color Bulb, see [example-full-colour](https://github.com/mwittig/node-milight-promise/blob/master/examples/example-full-colour.js#L41-L42). The command function in this case is `whiteTemperature(zone, percent)` where a relative percentage between lower and upper color temperature is specified as prescribed by the milight firmware.

mx-web, 2019-05-23:
`light.sendCommands(commands.fullColor.whiteTemperature(0, 100))` worked for me!

But does 100 stand for 6500K and 0 for 2700K?

mx-web, 2019-05-23:
light.sendCommands(commands.fullColor.whiteTemperature(0, 100)) worked for me!

closed 2019-05-23 by mx-web

mwittig, 2019-05-23:
That's great. Btw, the percent values roughly map to the following temperatures:

* 0% - 2700K (Warm White)
* 25% - 3650K
* 50% - 4600K
* 75%, - 5550K
100% - 6500K (Cool White)