mwittig/node-milight-promise #68: YL5 LED strip controller support

joeio, 2019-10-30:
It appears as if the [Milight YL5 LED strip controller](http://futlight.com/productdetails.aspx?id=347&typeid=151) should be supported by this package but I'm having problems getting it to work.

The discoverBridges function does discover the device as v6 with a name of "HF-LPT230" but no other commands I've tried work. I always get "no response timeout" errors.

The device is listening on UDP 5987 and on TCP 80, although I get a 404 not found trying to open the latter through HTTP.

I was also able to find that as soon as it's on wi-fi, it establishes a permanent connection to "cm2.xlink.cn", regardless if it's "activated" in the Milight Cloud app or not. From what I could gather, it would appear that both the YL5 and the Milight Cloud app establish a socket connection with XLink cloud and that the commands are sent and received through that socket. Apart from device discovery and wi-fi connection, everything else going on between the app and the device is done through the cloud but instead of just sending individual API commands through HTTP, it does it through the socket connection. Maybe all of this is already known but I thought I would mention what I was able to find.

Thoughts? Ideas?

TheLD6978, 2019-11-05:
After connecting the controller to wifi via the app, i can access its webserver. Its just a basic reconfiguration UI which allows you to configure a different ssid/password and lists the software version plus device name. That probably also gets used during the app configuration process.

If someone could shed some light on how the YL5 communicates through the websocket that would be awesome!

joeio, 2019-11-06:
Could you please share the path to the web UI you're seeing? The only address I was able to get something other than a 404 was http://X.X.X.X/iweb.html which allows one to upgrade the firmware of the LPT230 chip.

I did a lot of Wiresharking but my limited experience didn't get me really far. It appears to be a back and forth HEX payload which is probably similar to the one used on V6 protocol but I haven't been able to make any correlation.

Phil1pp, 2019-11-17:
Did you have any progress on this?
Did you try using the LimitlessLEDv6 tool?
[https://www.dropbox.com/s/pqepq1vereagpin/LimitlessLEDv6.exe?dl=0](url)

I am able to red and change all the settings.
However I had not luck on sending any light commands.
By using "set cloud server" and "TCP,38899,PC_IP" the YL5 was establishing a socket connection to my test server.
Now we just need to figure out the correct LED commands.

There is also one App on google play "com.epocapp.milightcontroller" that is supposed to work without cloud access.
When I set it v6 Box and CCT/RGBWW port 8899, and open the controls my YL5 flickers and then turns red.

djsomi, 2019-12-30:
Hey guys, did you manage to find anything about these controllers? Would be nice to implement into homebridge.

Thanks!

buenyamin-olgun, 2020-01-09:
> Hey guys, did you manage to find anything about these controllers? Would be nice to implement into homebridge.
> 
> Thanks!

I'm using the LS2 model which looks very similiar but does not have the WiFi functionality integrated.
Currently it runs with this homebridge plugin: https://github.com/buenyamin-olgun/homebridge-milight-v6

For the WiFi functions I previously had a iBox2 running, now replaced it with a [ESP8266 hub](https://github.com/sidoh/esp8266_milight_hub)

With the iBox2 I can call Port 80, I can call /iweb.html and I can read data with the LimitlessLEDv6 tool but I can't set any colour with the tool! Instead I've tested the 8zone-v6 script which is working.

@joeio 
Could you test https://github.com/mwittig/node-milight-promise/blob/master/examples/example-8zone-v6.js

mwittig, 2020-01-14:
Thanks @buenyamin-olgun for stepping in. From reading the link included in the first post my understanding is as follows: The product contains a V6 bridge which can be controlled via UDP 5987. The product is claimed to be compatible with all Might remotes. Thus it should also with with fullColor (RGB+CCT) control of node-milight-promise set which is basically the command set of the FUT092 remote. 

Important notice. The bridge needs to be paired/linked to the type of remote control you want to use. The recommended way of doing this is to use the Smartphone app. Besides the 8-channel remote I would expect the RGB+CCT (FUT092) remote to work. 

@joeio When you are able to control the lights with the 8-channel remote on you Smartphone the example-8zone-v6.js scripts should also work as suggested by @buenyamin-olgun You may need to change the IP address used in the script, however. See notice here: https://github.com/mwittig/node-milight-promise/blob/master/examples/example-rgbw-v6.js

Phil1pp, 2020-01-14:
I was able to analyze the commands using the Milight app on an android emulator and wireshark.
The YL-5 can be controlled only by UDP commands on port 5987.
The commands are different than on the other limitless/milight devices.

Before you can send any control commands you have to get a "session id" from the device.
I made a C# app to control on/off, brightness, color temperature and send individual commands.

I will release the source within the next few days.

mwittig, 2020-01-26:
That's great. However, I am wondering whether or not you have paired the device RGB+CCT (FUT092) remote as suggested earlier. If it is paired with the FUT092 I expect it will work with the existing code (using fullColor) given that the product works as advertised.

kukelivan, 2020-04-17:
@Phil1pp succeded? can you lay out an example code?

Phil1pp, 2020-10-26:
Sorry for the long delay.
I finally managed to upload some sample code.
[https://github.com/Phil1pp/MiLightYL5](url)

Enter IP address of controller, open socket and then pair.
Once you are successfully paired and a session id is displayed, you can execute the other commands.
Adjust brightness, color, saturation and temperature.

You can also send HEX commands.

ON
41,00,00,80,06,01,00,00,00,80,80
OFF
41,00,00,80,06,02,00,00,00,80,80

Pulse Up, Down (Brightness adjusted seperately)
41,00,00,80,05,01,00,00,00,80,80
Flashing (Brightness adjusted seperately)
41,00,00,80,05,02,00,00,00,80,80
Pulse Up, Down and 3 flashes (Brightness adjusted seperately)
41,00,00,80,05,03,00,00,00,80,80
Nightlight on
41,00,00,80,05,10,00,00,00,80,80
Default mode (Nightlight off)
41,00,00,80,06,06,00,00,00,80,80

Decrease Pulse Speed
41,00,00,80,06,03,00,00,00,80,80
Increase Pulse Speed
41,00,00,80,06,04,00,00,00,80,80

Color XX (0-255), repeat 4 times
41,00,00,80,01,XX,XX,XX,XX,80,80
Brightness, XX percent to hex
41,00,00,80,02,XX,00,00,00,80,80
Colortemperature CCW, XX 0=2700K, 0x26 (38dec)=6500K
41,00,00,80,03,XX,00,00,00,80,80
Saturation, XX percent to hex
41,00,00,80,04,XX,00,00,00,80,80

djsomi, 2021-06-20:
Hey guys, is there any progress on this?